BIPES/BIPES #10: Old MicroPython tools don't work on newer versions of Micropython

ptal78, 2021-06-26:
Examples:
'machine.RTC.now()' doesn't work and it should be RTC().datetime() and it returns a tuple

...

rafaelaroca, 2021-06-27:
It is really a bug, but not for all boards. 

For ESP8266 we have to use datetime(). Just tested and it works.

>>> os.uname()                                                                                                          
(sysname='esp8266', nodename='esp8266', release='2.0.0(5a875ba)', version='v1.15 on 2021-04-18', machine='ESP module wit
h ESP8266')          

The problem is with ESP32, when we have to use now()

>>> import os                                                                                                           
>>> os.uname()                                                                                                          
(sysname='esp32_LoBo', nodename='esp32_LoBo', release='3.2.24', version='ESP32_LoBo_v3.2.24 on 2018-09-06', machine='ESP
32&psRAM board with ESP32') 
                                                                                                   
We'll try to fix it tomorrow by generating different RTC calls for ESP8266 and ESP32.

Thanks for reporting these errors. Please feel free to keep contributing and helping BIPES project!

rafaelaroca, 2021-06-27:
Sorry, you are correct here. Tried to flash a recent version of MicroPython for ESP32 and it also uses datetime() instead of now()

ptal78, 2021-06-27:
I also had to put the brackets after the RTC such as:

RTC().datetime()
and
RTC().init()

to get the code to work in MU editor.

closed 2021-07-02 by ptal78