BIPES/BIPES #9: code malufunctions on machine. tools with duplicate function name

ptal78, 2021-06-26:
RTC.now and many other tools generate python scripts with a duplicate function such as RTC.RTC as shown below. This happens for almost all of the 'machine.' tools.

oled.text(('Year = ' + str((machine.RTC.RTC.now()
)[0])), 0, 7)

Also, notice the code goes on to a second line for no reason.

rafaelaroca, 2021-06-27:
Sorry for that. I guess you may have used the option from the last toolbox item (micropython), right?

All these blocks were automatically generated by a script that reads MicroPython documentation and generate the blocks, so several of these blocks were not tested.

The blocks outside micropython category were "hand written" and better tested. For example, in the timing category there are working RTC blocks.

The second line is a also from the script that adds a \n to each block command. We will work on fixing the new line and the code generated. 

Anyway, for your RTC issue, does the option in the timing category solves your issue?

ptal78, 2021-07-02:
Thanks for all your work. RTC in the timing category now works well.

closed 2021-07-13 by JorgeGMarques