BIPES/BIPES #13: New RTC method in Timing init produces no code

ptal78, 2021-06-27:
To get the RTC to initialize with code there is an RTC init tool in the Timing folder to accept 6 numerical values but it produces no code.

It should produce code like the following (and please notice the double brackets):

rtc.init((2021, 6, 28, 0, 8, 40, 0, 0))  # year, month, date, 0, hours, minutes, seconds, 0

JorgeGMarques, 2021-06-27:
I'll take a look into it.

Thanks!

rafaelaroca, 2021-06-28:
Hi Jorge and ptal78. 

The code generator for "Set RTC" was empty. Just implemented and tested it, but I need to discuss something with you.

I tested the implementation on 3 boards.
A. ESP8266 (MicroPython v1.12 on 2019-12-20; ESP module with ESP8266)                                                              
B. ESP32 (MicroPython v1.12-535-g509359754 on 2020-06-13; ESP32 module with ESP32 )
C. ESP32 (MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32&psRAM board with ESP32) 

In A and B cases, rtc.datetime() works both for reading and writing the RTC.
In the C case, there is no rtc.datetime(), so I implemented as rtc.init() and rtc.now(). What is interesting is that case B is also for ESP32, but does not support now(), but datetime()!

By the way, NTP works nicely! You can set your RTC from the Internet using the NTP block. Working also.

Anyway, it is working. Could you please try?

thanks
Rafael

rafaelaroca, 2021-06-28:
Ok, tested again:
MicroPython v1.15 on 2021-04-18; ESP module with ESP8266
>>> rtc.datetime()

MicroPython v1.16 on 2021-06-23; ESP32 module with ESP32                        
>>> rtc.datetime()                                                              

So, in newer versions, both have rtc.datetime()

ptal78, which version do you have? Is there a datetime() function?

thanks

ptal78, 2021-06-28:
Thanks,

Yes the rtc.datetime() works to both read and write to the RTC on
micropython v1.16

rtc.init() works but just to set the time.

Regards,
Phil

ptal78, 2021-06-28:
I've tested the output of the Tool called Set RTC Value and although it still doesn't work I just wrote the correct code below in the Tool 'Run Python Code' and this Python code below works:

rtc.init((2021, 6, 28, 0, 8, 40, 0, 0))

the following also works

rtc.datetime((2021, 6, 28, 0, 8, 40, 0, 0))

When generating the code from your Set RTC Value tool,
just make sure you use double brackets to indicate a tuple.
And also make sure there are 8 values, with the fourth and last also generated as zero.

rafaelaroca, 2021-06-28:
Hi. I did not understand, sorry. you say the generated code is correct, but it does not work. Could you please reload the page / refresh with Shift+F5 or Ctrl+F5 to be sure the latest version is used. Just tested here and worked.

ptal78, 2021-06-28:
What Blocks are you using to set the RTC?

I only have it working using the Python Code block.

ptal78, 2021-06-28:
Sorry. Checked again and it's working now.

Thanks

ptal78, 2021-06-28:
But you changed Get RTC tool and now it is not working.

ptal78, 2021-06-28:
Sorry they are both working now.

But Get RTC Value Block has been changed to Get RTC Value (ESP8266) for
some reason. But it still works on the ESP32.

ptal78, 2021-06-28:
The code below works now thanks, but not sure why both GetRTC and SetRTC blocks now say ESP8266 when it clearly works for the ESP32 also.
[image]

closed 2021-07-02 by ptal78