BIPES/BIPES #15: Display block for ST7789 LCD

ptal78, 2021-06-30:
Hi,

Can you possibly include a display driver for the ST7789?

Phil

JorgeGMarques, 2021-06-30:
Hi,

do you have a suggestion of a micropython library with the driver?

Best regards,

ptal78, 2021-06-30:
Hi,

Yes, ST7789py.py file here:
https://github.com/devbis/st7789py_mpy

Regards,

*Philip Tallents*

*Manager*
*PicoKit*
ABN: 88 995 830 437
Ph: 0402 239 363
Ph: (07) 5330 3095
Fax: (07) 3465 0154

rafaelaroca, 2021-07-27:
We have added a basic block for the library you mentioned. It is not fully implemented, but already works on boards like this:
https://pt.aliexpress.com/item/33048962331.html
(TTGO ESP32 with ST7789 display)
Pins are still fixed, but the basics are working.
We started trying to implement text printing, but the library is very slow! 

Hope we advance here soon.

rafaelaroca, 2021-08-29:
Basic support is now functional, including configurable adjustable pins, colors, line and pixel drawing. Soon, hope to add text support. 

https://github.com/BIPES/BIPES/commit/c72593caf6fe3fb70d9ff1d6d166b654b79bc75e

closed 2021-08-29 by rafaelaroca