BIPES/BIPES #16: NTP fails with a traceback of 3 errors

ptal78, 2021-07-02:
I'm not sure why, but the NTP doesn't seem to work. I get the following traceback info:

Traceback (most recent call last):                                                                                      
  File "<stdin>", line 6, in <module>                                                                                   
  File "ntptime.py", line 34, in settime                                                                                
  File "ntptime.py", line 20, in time                                                                                   
OSError: -202

rafaelaroca, 2021-07-02:
Hello! Are you connected to the Internet? I ask that because I just tested it, and worked perfectly for me on ESP8266. 
OSError: -202 may be related to Internet connection unavailable.

ptal78, 2021-07-17:
Does the ESP32 have to be connected to WiFi to enable this?

rafaelaroca, 2021-07-17:
Yes! NTP stands for Network Time Protocol. It gets the current date and time from NTP servers over the Internet, some connected to atomic clocks or GPS modules. It is a very precise system to update your system clock using the Internet. If you just want to have local time, without the need for the Internet, you can simply use RTC blocks. There is one to set up the ESP32 RTC and others to read. If you are not connected, you will get an error when running code with this block. This is why I think this issue is not a bug :-)

rafaelaroca, 2021-07-31:
Hi! I really think this is not a bug. NTP only works when connected. Can we close this bug? @ptal78 would you like to add any comments?

rafaelaroca, 2021-08-18:
Closing the issue, as it is not a bug, IMHO. Will only work with an active network connection. You can use the RTC block to keep the time and get timing information.

closed 2021-08-18 by rafaelaroca