BIPES/BIPES #19: Feature request - import neopixel - and custom Neopixel Display block

ptal78, 2021-07-03:
Neopixel (ws2812) LED RGB strip lights protocol is already included in micropython for multicolored displays.

Example user code:
>>> import machine, neopixel
>>> np = neopixel.NeoPixel(machine.Pin(4), 8)

>>> np[0] = (255, 0, 0) # set LED 0 to red, full brightness
>>> np[1] = (0, 128, 0) # set LED 1 to green, half brightness
>>> np[2] = (0, 0, 64)  # set LED 2 to blue, quarter brightness

These could be easily configured with three blocks
- **Neopixel init** block to define the pin and number of LEDs, and Name of the Neopixel to produce the following code:
>>> import machine, neopixel
>>> np = neopixel.NeoPixel(machine.Pin(4), 16)

- **Neopixel** block that selects the name of the Init and then has horizontal space for a number of variables to produce the code for as many variables are loaded into the block and according to their position. eg, a _variable_ Color0 in position 2 (of 16) would only produce the following code:
>>> np[1] = Color0

- **Color** block to produce a single tuple variable with (3 numbers) for example ** Color0 = (R, G, B)**. This block has a round circle to display a single color as selected from a color lookup chart. These Colour blocks can be copied and placed like other variable blocks. By themselves, they would generate code for example a red LED on half brightness:
- Color0 = (128, 0, 0)
- copying Color blocks doesn't always create new variables, until the color is changed

If you need more thoughts about this let me know.

ptal78, 2021-07-05:
Oh, I forgot the following:

np.write()  # required to transmit the data from the buffer array to the LEDs

Also, if you would like an example of the code, the following implementation is much preferred than the older ESP8266 documentation does it (and should still work for both).

ESP32:     https://docs.micropython.org/en/latest/esp32/quickref.html?highlight=esp32%20neopixel

This also has an explanation of the APA106 driver, and the Touchpad driver.

rafaelaroca, 2021-07-05:
Hello!
Done, but not tested. Took the opportunity to write about how to create new blocks:
https://bipes.net.br/wp/creating-new-blocks/

Example:
https://bipes.net.br/beta2serial/ui/#i3dbcf

Could you test it please?

Only implemented in the beta2serial. If it works, we can copy the code to WebRPL (network version)

best regards!
Rafael

ptal78, 2021-07-10:
Excellent - it works great! You are a champion.

rafaelaroca, 2021-07-11:
:-)

closed 2021-07-11 by rafaelaroca

lbaitemple, 2023-03-09:
> :-)

c

> Hello! Done, but not tested. Took the opportunity to write about how to create new blocks: https://bipes.net.br/wp/creating-new-blocks/
>
> Example: https://bipes.net.br/beta2serial/ui/#i3dbcf
>
> Could [...]

Can you make the neopixel work in raspberry pi pico? I tried on offline version, but I cannot include the neopixel library.

jetannenbaum, 2023-03-09:
The offline version has been updated to use the internal MicroPython neopixel library for the Pico.  You need to use version 1.18 or later of the MicroPython firmware.  Available here: https://micropython.org/download/rp2-pico/

lbaitemple, 2023-03-09:
> 
Thanks. I tried that, but the code does not run. The follow error appears on 
AttributeError: 'module' object has no attribute 'NeoPixel'

I think the problem is the NeoPixel was declared incorrectly in the code. I guess we cannot use the block code to control the NeoPixel.

jetannenbaum, 2023-03-10:
Hi,

   I'm a bit confused.  I tried the neopixel blocks in both the "standard" bipes code (http://bipes.net.br/ide/) and a branch that is specific to the RP2040 boards (https://bipes.net.br/pico/ui/).  In both cases, the neopixel blocks ran without error.  I then attempted to run the downloaded code from https://bipes.net.br/wp/  That version of code didn't have the capability to use a neopixel display with the Pico.  Finally, I tried the code from the offline branch.  Once I built the index_offline file and tried the neopixel block, it ran fine.  Can you supply a few more details on which code you were running?  Thanks

lbaitemple, 2023-03-10:
> Hi,
>
> I'm a bit confused. I tried the neopixel blocks in both the "standard" bipes code (http://bipes.net.br/ide/) and a branch that is specific to the RP2040 boards (https://bipes.net.br/pico/ui/). [...]

Thank you. I figured it out. I was using a wrong neopixel python file (https://github.com/blaz-r/pi_pico_neopixel/blob/main/neopixel.py). I uploaded micropython neopixel python file. It runs without any error, but I would hope we can change the brightness of the neopixel display. Thank you again for your quick response.

zegonix, 2023-09-14:
Hi,
I tried to use the NeoPixel library with my Raspberry Pi Pico. I noticed, that the blocks are not listed when working with the Pico, but are when working with the ESP32.
I tried to generate a little test program for the ESP32 and then switch to the Pico, but I can't test it, as it doesn't connect (I use Chrome with 'experimental wep platform features' enabled). The website doesn't support https and this [article](https://rafaelaroca.wordpress.com/2020/06/21/access-serial-devices-directly-from-the-browser/) says that the web terminal only work with https, so maybe this is the problem? 
I will try it offline, but since we plan to use this in a workshop with kids, the offline version would be a dealbreaker for us.