BIPES/BIPES #22: Urgent - PWM code not working on ESP32 due to incorrect use of machine.Pin

ptal78, 2021-07-10:
The code below is generated with the PWM tool which doesn't work due to use of machine.PWM and of machine.Pin.

Resolve by removing both strings 'machine.' at line 4

Corrected code should not use 'machine.' before the function. PWM(... is called. PWM should be called without the preceding 'machine.' 
Similarly, 'machine.' should not be included before 'Pin' 

1     from machine import Pin
2     from machine import PWM
3
4     pwm14 = machine.PWM(machine.Pin(14))
5
6
7     pwm14.freq(1000)
8     pwm14.duty_u16(50)

ptal78, 2021-07-10:
line 4 should simply be:
`pwm14 = PWM(Pin(14))`

lines 1 & 2 could be combined as:
`from machine import Pin, PWM`

ptal78, 2021-07-10:
Also line 8 has an error in it and should rather say:
`pwm14.duty(50)`

rafaelaroca, 2021-07-11:
Solved. Could you just please reload and check if it is ok for you?

ptal78, 2021-07-12:
Thanks so much.

Yes it works now.

closed 2021-07-12 by ptal78