BIPES/BIPES #23: ADC not working in ESP32, and new tool suggestions for better ADC control

ptal78, 2021-07-11:
Code generated with ADC pin is wrong
    import machine
    adc(35)=machine.ADC((35))

Correct code should be:
    from machine import ADC, Pin
    adc35 = ADC(Pin(35))

Also, please include tools to generate the following lines:

adc.atten(ADC.ATTN_0DB)    # options are ATTN_0DB, ATTN_2_5DB, ATTN_6DB, ATTN_11DB
adc.width(ADC.WIDTH_10BIT)   # options are WIDTH_9BIT, WIDTH_10BIT, WIDTH_11BIT, WIDTH_12BIT

rafaelaroca, 2021-07-11:
Done :-) 
Could you please check if it is ok? 
Thank you very much for helping testing all the blocks!

ptal78, 2021-07-12:
No, still not working.=== import machine    

try changing this line:
     adc35=machine.ADC(35)

to this line:
     adc35 = machine.ADC(machine.Pin(35))

But it would be faster code if you do the import the new way the micropython suggests
And change this line:
     import machine

To this line:
     from machine import Pin, ADC

And:
     adc35=machine.ADC(35)

To:
     adc35 = ADC(Pin(35))
 

Error code displayed below:                                                                                          
===                                                                                                                     
=== Ultra = None                                                                                                        
===                                                                                                                     
=== adc35=machine.ADC(35)                                                                                               
===                                                                                                                     
===                                                                                                                     
=== Ultra = adc35.read()                                                                                                
===                                                                                                                     
===                                                                                                                     
Traceback (most recent call last):                                                                                      
  File "<stdin>", line 5, in <module>                                                                                   
ValueError: expecting a pin

rafaelaroca, 2021-07-12:
Solved. Could you please check if it works? Thanks!

ptal78, 2021-07-17:
Thanks.

It seems to work well when set to Attn_11DB for 3.3V operation.

You might want to make this the default value instead of Attn_0DB, which didn't work for my setup at all.

Kind Regards,
Phil

ptal78, 2021-07-17:
It works except, the ".read_u16()" causes a significant error.

Please change this to ".read()"

JorgeGMarques, 2021-07-21:
Commit cbf94e1  changes  `read_u16()` this to `read()`.

closed 2021-08-09 by JorgeGMarques