BIPES/BIPES #26: Downloading Blocks from ESP32 to xml file works but uploading doesn't

ptal78, 2021-07-12:
Is there something that can be done about this problem?

JorgeGMarques, 2021-07-12:
Can you provide the XML file that you are trying to upload so I can debug?

Thanks in advance

JorgeGMarques, 2021-07-12:
I'll look into that

rafaelaroca, 2021-07-12:
Hello! Just to help, this feature seems ok to me. Tested a simple program on serial and network versions, and they seem ok....but I will wait for Jorge's opinion.

JorgeGMarques, 2021-07-12:
@ptal78 could you provide a misbehaving XML? Couldn't find the bug here (eg.: https://bipes.net.br/beta2serial/ui/#vg9o2h works fine)

ptal78, 2021-07-17:
Sorry for the delay.
Here is code for one of my XML files. I actually can't get any of my XML files to upload into Bipes Blocks. And this github won't upload XML script files.

```
<xml xmlns="https://bipes.net.br">
  <workspace>
    <field name="DEVICE">ESP32</field>
    <field name="TIMESTAMP">1626498992080</field>
  </workspace>
  <variables>
    <variable id="@k6Y#w~7^.hAlcq)n=N6">LDR_value</variable>
  </variables>
  <block type="controls_whileUntil" id="6xm(9uos7MfV7q:EJ-4c" x="363" y="-138">
    <field name="MODE">WHILE</field>
    <value name="BOOL">
      <block type="logic_boolean" id="r;;w-Y78rx=.[{G_(m#5">
        <field name="BOOL">TRUE</field>
      </block>
    </value>
    <statement name="DO">
      <block type="variables_set" id="yXaQXr4?3h$=]9rv-k%(">
        <field name="VAR" id="@k6Y#w~7^.hAlcq)n=N6">LDR_value</field>
        <value name="VALUE">
          <block type="math_constrain" id="s7F0A=`Z6WspS])OlH=T">
            <value name="VALUE">
              <shadow xmlns="https://bipes.net.br" type="math_number" id="{E$OriBq0FZ~NYggzAHx">
                <field name="NUM">50</field>
              </shadow>
              <block type="math_arithmetic" id=",{k[^WjU|$S0[uJO,T^O">
                <field name="OP">MULTIPLY</field>
                <value name="A">
                  <shadow xmlns="https://bipes.net.br" type="math_number" id="x-:qOh%W$bM}X,BG=r~x">
                    <field name="NUM">1</field>
                  </shadow>
                  <block type="math_arithmetic" id="uH}gwF@UO(=r`SFyC%vh">
                    <field name="OP">MINUS</field>
                    <value name="A">
                      <shadow xmlns="https://bipes.net.br" type="math_number" id="!uAv3}hd0x}Ya?kd-677">
                        <field name="NUM">1</field>
                      </shadow>
                      <block type="esp32_adc" id="[$uvF$nDf:5$Q0m~;EUo">
                        <field name="Attenuation">3</field>
                        <field name="Width: ">1</field>
                        <value name="pin">
                          <shadow type="pinout" id="mH@CEWvICa:?IY2ME7t;">
                            <field name="DEVICE">ESP32</field>
                            <field name="PIN">35</field>
                          </shadow>
                        </value>
                      </block>
                    </value>
                    <value name="B">
                      <shadow type="math_number" id="A.ib!%Y]%LHGt.h%DqM/">
                        <field name="NUM">580</field>
                      </shadow>
                    </value>
                  </block>
                </value>
                <value name="B">
                  <shadow type="math_number" id="*tk!=@M]BO3zCONXeb8$">
                    <field name="NUM">5</field>
                  </shadow>
                </value>
              </block>
            </value>
            <value name="LOW">
              <shadow type="math_number" id="N!jP=-{qd=P)y#E6*VKs">
                <field name="NUM">1</field>
              </shadow>
            </value>
            <value name="HIGH">
              <shadow type="math_number" id="JXO;d3B@[teZxXGNXB=Z">
                <field name="NUM">1023</field>
              </shadow>
            </value>
          </block>
        </value>
        <next>
          <block type="delay" id="[XCjwUxPd~+ah(q[B;*.">
            <value name="time">
              <shadow type="math_number" id="{KRcTQj:#e/.A0Q7?(g/">
                <field name="NUM">0.1</field>
              </shadow>
            </value>
            <next>
              <block type="esp32_pwm" id="XA3f=fE0Su2F@]Dd)V$K">
                <field name="MSG_GET_MS">ESP32 PWM</field>
                <value name="pin">
                  <shadow type="pinout" id="A9Kl@q$Qe;$~7uIL-t-g">
                    <field name="DEVICE">ESP32</field>
                    <field name="PIN">14</field>
                  </shadow>
                </value>
                <value name="frequency">
                  <shadow type="math_number" id=".r3bnX;V4ZsXHO9If8RQ">
                    <field name="NUM">1000</field>
                  </shadow>
                </value>
                <value name="duty">
                  <shadow xmlns="https://bipes.net.br" type="math_number" id="fA5,7cRID05^yejqV){,">
                    <field name="NUM">512</field>
                  </shadow>
                  <block type="variables_get" id="0/LW42g?90_GOLF/8(Yq">
                    <field name="VAR" id="@k6Y#w~7^.hAlcq)n=N6">LDR_value</field>
                  </block>
                </value>
              </block>
            </next>
          </block>
        </next>
      </block>
    </statement>
  </block>
</xml>
```

ptal78, 2021-07-17:
Can you also make it prompt a save file dialog box so we can browse to a specific file location, and save it as a specific name?

ptal78, 2021-07-17:
This one is another file that won't load. Maybe it even has more simple information than the last.

```
<xml xmlns="https://bipes.net.br">
  <workspace>
    <field name="DEVICE">ESP32</field>
    <field name="TIMESTAMP">1626529243235</field>
  </workspace>
  <block type="controls_whileUntil" id="Qe1F-tbGp$qapEwE|+KB" x="138" y="162">
    <field name="MODE">WHILE</field>
    <value name="BOOL">
      <block type="logic_boolean" id="5N8[eON!C~#za()N,Qrd">
        <field name="BOOL">TRUE</field>
      </block>
    </value>
    <statement name="DO">
      <block type="controls_whileUntil" id="|7,v4c:J!1gk-I9zp67q">
        <field name="MODE">WHILE</field>
        <value name="BOOL">
          <block type="gpio_get" id=".v!Cp$*5.%Goqf|M_8E*">
            <value name="pin">
              <shadow type="pinout" id="bCAGu{|2~y(J*uU9MpQJ">
                <field name="DEVICE">ESP32</field>
                <field name="PIN">35</field>
              </shadow>
            </value>
          </block>
        </value>
        <statement name="DO">
          <block type="gpio_set" id="%K,1=Vo5IbIsBy.lyu@J">
            <value name="pin">
              <shadow type="pinout" id="TG|oMelE76tx-Z@N+F!r">
                <field name="DEVICE">ESP32</field>
                <field name="PIN">14</field>
              </shadow>
            </value>
            <value name="value">
              <shadow type="logic_boolean" id="u}B37kekNb:e_K+xM{%1">
                <field name="BOOL">FALSE</field>
              </shadow>
            </value>
          </block>
        </statement>
        <next>
          <block type="controls_whileUntil" id="@xc{/AH1oT81rjv+rKQ}">
            <field name="MODE">WHILE</field>
            <value name="BOOL">
              <block type="gpio_get" id="Z]x|6h+q9+_Z_c#{h%FN">
                <value name="pin">
                  <shadow type="pinout" id="q:3y**Xm=GVH.u9^bCt@">
                    <field name="DEVICE">ESP32</field>
                    <field name="PIN">34</field>
                  </shadow>
                </value>
              </block>
            </value>
            <statement name="DO">
              <block type="gpio_set" id="[T(y:a`:[wNXo--)v?-@">
                <value name="pin">
                  <shadow type="pinout" id="$XH/:a[S)knuU(,uO0QL">
                    <field name="DEVICE">ESP32</field>
                    <field name="PIN">14</field>
                  </shadow>
                </value>
                <value name="value">
                  <shadow type="logic_boolean" id="nHHs[^^%e`^KIa_/K-.F">
                    <field name="BOOL">TRUE</field>
                  </shadow>
                </value>
              </block>
            </statement>
          </block>
        </next>
      </block>
    </statement>
  </block>
</xml>
```

ptal78, 2021-07-23:
I still can't get this upload button to do anything except open the dialog box to select the file. Once selected, click open, and nothing happens.

[image]

Here's a very simple xml file with a single NeoPixel block to show nothing opens.

```
<xml xmlns="https://bipes.net.br">
  <workspace>
    <field name="DEVICE">ESP32</field>
    <field name="TIMESTAMP">1627077286544</field>
  </workspace>
  <block type="neopixel_init" id="bD+6:[U2E)04wk-Ra)nM" x="412" y="88">
    <value name="pin">
      <shadow type="pinout" id="ow{%8`|HIkiXqR8e`NPk">
        <field name="DEVICE">ESP32</field>
        <field name="PIN">36</field>
      </shadow>
    </value>
    <value name="number">
      <shadow type="math_number" id="pzqX+LrMy2+2$.uL|~9L">
        <field name="NUM">8</field>
      </shadow>
    </value>
  </block>
</xml>
```

JorgeGMarques, 2021-07-24:
Hi!
I need your help on this, since I can't replicate the bug!
- What's the OS and browser you are using?
- When you try to load does some error appear in the developer console `CTRL+SHIFT+I`?

Here are two quick videos on how to use the developer console in [firefox](https://youtu.be/TmJVGZz7cgY?t=212) and [chrome](https://www.youtube.com/watch?v=H0XScE08hy8)

I was able to get it to fail sometimes loading on Windows/Chrome and my developer console returned this error:
```
ui.js:270 Uncaught (in promise) TypeError: Failed to execute 'readAsText' on 'FileReader': parameter 1 is not of type 'Blob'.
    at ui.js:270
```
If you are getting some error in the developer console, please provide so I can pinpoint the source of the bug!

Regards,

ptal78, 2021-07-25:
Hi,

Here's the error in chrome from the DevTools window:

```
ui.js:260 Uncaught (in promise) TypeError: Failed to execute 'readAsText' on 'FileReader': parameter 1 is not of type 'Blob'.
    at ui.js:260
```

ptal78, 2021-07-25:
It also says the following when I expand the arrow:
```

<html>
<body>
<!--StartFragment--><div tabindex="-1" class="console-group console-group-messages" style="box-sizing: border-box; min-width: 0px; min-height: 0px; position: relative; color: rgb(34, 34, 34); font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div tabindex="-1" class="console-message-wrapper console-error-level console-selected" style="box-sizing: border-box; min-width: 0px; min-height: 0px; outline-width: 0px; display: flex; border-top-width: ; border-top-style: ; border-top-color: rgb(204, 222, 245); border-bottom-width: 1px; border-bottom-style: solid; border-bottom-color: var(--error-border-color); background-color: rgb(236, 241, 248); --error-text-color: rgb(200 0 0);"><div class="console-message" style="box-sizing: border-box; min-width: 0px; min-height: 17px; clear: right; position: relative; padding: 3px 22px 1px 0px; margin-left: 24px; flex: 1 1 auto; display: flex;"><div class="console-message-stack-trace-toggle" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 1 1 auto; display: flex; flex-direction: row; align-items: flex-start;"><div class="console-message-stack-trace-wrapper" role="tree" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 1 1 auto; display: flex; flex-direction: column; align-items: stretch;"><div class="" role="group" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 0 0 auto;"><span class="monospace" style="display: inline-block; --monospace-font-size: 12px; --monospace-font-family: consolas, lucida console, courier new, monospace; --source-code-font-size: 12px; --source-code-font-family: consolas, lucida console, courier new, monospace; box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--monospace-font-family); font-size: var(--monospace-font-size)  !important;">

(anonymous) | @ | ui.js:260
-- | -- | --
  | Promise.then (async) |   |  
  | workspace.loadXML | @ | ui.js:258
  | loadButton.onclick | @ | ui.js:175

</span></div></div></div></div></div></div><div aria-hidden="true" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; height: 0px; color: transparent;">﻿</div><div class="source-code" id="console-prompt" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; white-space: pre-wrap; font-size: 12px; clear: right; position: relative; margin: 0px 22px 0px 20px; color: rgb(34, 34, 34); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="widget" style="box-sizing: border-box; min-width: 0px; min-height: 0px; position: relative; flex: 1 1 auto; contain: style;"><div class="console-prompt-editor-container" style="box-sizing: border-box; min-width: 0px; min-height: 21px;"><div class="widget vbox" tabindex="-1" style="box-sizing: border-box; min-width: 0px; min-height: 0px; position: relative; flex: 1 1 auto; contain: style; display: flex; flex-direction: column !important; overflow: hidden;"><div class="CodeMirror cm-s-default CodeMirror-wrap source-code fill" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--source-code-font-family); white-space: pre-wrap; font-size: var(--source-code-font-size)  !important; position: relative; inset: 0px; overflow: hidden; background-image: ; background-position-x: ; background-position-y: ; background-size: ; background-repeat-x: ; background-repeat-y: ; background-attachment: ; background-origin: ; background-clip: ; background-color: transparent !important; color: rgb(34, 34, 34); height: auto; line-height: 1.2em !important; padding: 3px 0px 1px; width: 512px;"><div style="box-sizing: border-box; min-width: 0px; min-height: 0px; overflow: hidden; position: relative; width: 3px; height: 0px; top: 3px; left: 4px;"><textarea autocorrect="off" autocapitalize="off" spellcheck="false" aria-label="Console prompt" tabindex="0" style="box-sizing: border-box; min-width: 0px; min-height: 0px; resize: none; overflow: hidden; position: absolute; bottom: -1em; padding: 0px; width: 1000px; height: 1em; outline: none;"></textarea></div><div [... cut ...] nherit; color: inherit; z-index: 2; position: relative; overflow: visible; -webkit-tap-highlight-color: transparent; font-variant-ligatures: contextual; word-break: normal;"><span role="presentation" style="box-sizing: border-box; min-width: 0px; min-height: 0px; padding-right: 0.1px;"><span cm-text="" style="box-sizing: border-box; min-width: 0px; min-height: 0px;">​</span></span></pre></div></div></div></div></div></div></div></div></div></div></div></div><!--EndFragment-->
</body>
</html>
```

ptal78, 2021-07-25:
It seems to be occurring at the following location *****:

```
workspace.prototype.loadXML = function () {
  this.promptFile().then((file) => {
	  let reader = new FileReader ();
********  reader.readAsText(file,'UTF-8');  ********
    let self = this;
```

ptal78, 2021-08-09:
Did you find a fix? What browser are you using?

ptal78, 2021-08-09:
We are using Google Chrome Version 92.0.4515.131
And uploading xml files still isn't working for us.

We are considering that this feature is mandatory for our students to develop code and are running out of time (and patience).

JorgeGMarques, 2021-08-09:
Hi,
I added some types checking on commit 4267856, to avoid the function throwing a `TypeError`: 
```
if(typeof file == 'object' && /.xml$/.test(file.name) && file.type == 'text/xml'){...}
```

hope it works now.

ptal78, 2021-08-09:
Now it just pops up a message saying "No valid file selected to load"

ptal78, 2021-08-09:
Can you post here a valid file that works for you and the version of web browser you are using?

ptal78, 2021-08-09:
I just used Chrome on my personal laptop Version 92.0.4515.131 (Official Build) (64-bit) and it uploaded fine.

This is so odd that it works with the exact same browser at home but not on the work computers.

JorgeGMarques, 2021-08-09:
That's annoying... I will replace the whole thing later today with a solution I used to use in other projects.

JorgeGMarques, 2021-08-10:
I just rewrote the whole thing in  d16b0b489ae81eab612c7925327158e62b24be9f , please check if it works now.

If the problem persists, please consider checking if some software might be affecting how your browser is handling files, like a antivirus.

ptal78, 2021-08-10:
Amazing,

It works now.

You are a magician!

closed 2021-08-10 by JorgeGMarques