BIPES/BIPES #27: Feature idea - Blocks Code tabs to store multiple programs open at the same time

ptal78, 2021-07-17:
The idea came from a colleague. We would like to have multiple code examples open at the same time.

Pressing the program button loads the code of only one tab, that is currently displayed.

ptal78, 2021-07-17:
My colleague also suggested it could be instead like a cabinet icon on the right side of the screen, so we can drag existing code into the cabinet storage, much like you might drag it into the bin. Then clicking on the bin, we get a graphical display of each code we placed in the cabinet.

rafaelaroca, 2021-08-29:
Hi @ptal78  :-)

Does the new file manager created by @JorgeGMarques solve this issue?

[image]

JorgeGMarques, 2021-08-29:
It doesn't haha. It's about multiplex projects. I think you be a nice feature too.
Would use local storage to keep multiple project instead of only one.

JorgeGMarques, 2021-09-11:
Work in progress.
Finally working  on it, it's already function but still need to fill all the possibility, like fetching the "legacy" single project, so no one would loose their work.

[image]

rafaelaroca, 2021-09-12:
Wow! Looking forward to use it. Great! :-)

JorgeGMarques, 2021-09-15:
Done with 027de7d0d22a4ab80401ccfb28887a9b2689ab8e, e48b4c5e21672baef1371470a4da15460d876f44 and ce4971bed93b5b9df7714501dd24af110330bd83.
Shared projects are imported to the account panel and throws a notification.

closed 2021-09-15 by JorgeGMarques