BIPES/BIPES #29: NeoPixel need an np.write() block and please remove np.write() function from the NeoPixel block

ptal78, 2021-07-17:
The NeoPixel block is great fun, but it seems to send a lot of data because the np.write() is called every time the Neopixel block sets one LED color.

So with 32 LEDs, it sends out 32 times more data than it needs to, because you should be able to set the whole list of 32 LED colors and call the np.write() function just once.

Can you possibly create one more block for the NeoPixel and modify two that you already have?

1st block is new
`np.write()`

2nd and 3rd blocks to remove the function call np.write()
Currently:
```
np[0]=((255,0,0))
np.write()
```

Preffered:
`np[0]=((255,0,0))`

JorgeGMarques, 2021-07-20:
Done with 8b5173d, quick test: 
https://bipes.net.br/beta2serial/ui/#vi4mcm

Best regards,

closed 2021-07-20 by JorgeGMarques

ptal78, 2021-07-21:
Hi,

The double brackets will cause a problem.

[image]

Regards,
Phil