BIPES/BIPES #30: Python code fails to generate when NeoPixel color is set with a variable - please fix ASAP

ptal78, 2021-07-17:
I think the variables used to work with the NeoPixel, but now it doesn't.

[image]

From the image, everything works normally until the variable is placed in the 'color' block.
The python code fails to generate.

As you can see, the variable I set is a number, and as the number 123 is within the 0-255 required value, it should be acceptable within the color block. But it fails to generate any python code as soon as this variable is applied in either of the three variable locations.

Can you please fix this ASAP?

JorgeGMarques, 2021-07-20:
Hello,

The issue has been resolved, but keep in mind that `neopixel*` blocks are only available in the `beta2serial` version for now.
Quick test: https://bipes.net.br/beta2serial/ui/#tzqzae

The issue was due to the misuse of the `Blockly.JavaScript` functions. Since we use the Python generator,we should always invoke  `Blockly.Python` instead (both are not interchangeable, even though not always cause bugs).

Best regards,

ptal78, 2021-07-21:
Hi,

I'm not sure what's happening. You had this almost perfect 4 hours ago with the new neopixel write block and everything. Even the variables started producing code. 

All I said was you had double brackets, but now it's all screwed up and you've taken away the write neopixel block. Maybe you are just still working on it - are you? Because the main line of code to load the data is missing.

Where is this line?

np[0] = (255, 255, 255)

JorgeGMarques, 2021-07-21:
Hi,

please force refresh your browser cache, just committed cbf94e1 to remove the extra parenthesis and change the `ADC` block to `read()`, quick test: http://www.bipes.net.br/beta2/ui/#re4ih9

JorgeGMarques, 2021-07-21:
`XML` files are ridiculously persistent in modern browser, if "`CTRL + F5`" won't solve, please "`CTRL + F5`" these files:

`http://www.bipes.net.br/beta2/ui/toolbox_esp32.xml`
`http://www.bipes.net.br/beta2serial/ui/toolbox_esp32.xml`

ptal78, 2021-07-21:
Wonderful. I cleared my cache and it seems to works perfectly now.

Thankyou so much

closed 2021-07-22 by ptal78