BIPES/BIPES #34: NeoPixel Color defined simply by Hue and Brightness variables

ptal78, 2021-07-23:
Hi,

I have generated some code that nicely converts Hue and Bright variables into RGB for the neopixel as shown below:

Can you possibly put this code into a block with the other NeoPixel blocks before Monday?

Example Block Code that returns a RGB formatted list is shown within a print function.
It demonstrates two values representing Hue, and Brightness - both are values from 0 to 255

```
print( Hue_to_RGB(120, 50) )
```

Functions to include when the first "Hue to RGB" block is used:

```
import math

Hue = None
Bright = None

# Describe this function...
def Color(x):
  return (math.cos(((min(max(x, 0), 180) % 180) * 2) / 180.0 * math.pi) - 1) * -0.5 + 0.005

# Describe this function...
def Hue_to_RGB(y, z):
  return ((int((z * Color(y % 256)))),(int((z * Color((y + 85) % 256)))),(int((z * Color((y + 170) % 256)))))

```

The block could look like this below with two int or variable values ( Hue, and Brightness ). None of the variables need to be global.

[image]

ptal78, 2021-07-24:
Actually, Hue should really be from 0 to 360, so the Hue_to_RGB function needs to have the following code.

```
def Hue_to_RGB(y, z):
  return (int(z * Color((y + 120) % 360)), int(z * Color(y % 360)), int(z * Color((y + 240) % 360)))
```

And Color also needs to be changed to the following:
```
def Color(x):
  return (math.cos((min(max(x, 0), 240) * 1.5) / 180.0 * math.pi) - 1) * -0.5 + 0.0018
```

JorgeGMarques, 2021-07-24:
Hi, 
just committed the new `HSL_to_RGB` with a visual feedback to help set up the values.
quick test: https://bipes.net.br/beta2serial/ui/#6mpjxw 

[image]

I hope it fits you well,

closed 2021-07-24 by JorgeGMarques

ptal78, 2021-07-25:
Great work! Thanks so much.