BIPES/BIPES #35: Lists have a number minus 1 cause unexpected behaviour

ptal78, 2021-07-24:
In creating a list of colors for our NeoPixels, I have noticed the Bipes code generates a list number **'minus one'** for some odd reason. This is painful to have to teach students a code workaround because of a non-standard implementation of MicroPython in Bipes. We are trying to teach kids how to code in the MicroPython language, but Bipes is making this difficult by injecting non-standard code.

I am clearly getting the list value from a variable called LED and the list does clearly use the values from 0 to 7. Knowing this value in lists starts from 0 is standard MicroPython implementation.

[image]

But Bipes is clearly introducing a LED minus 1 calculation (as shown below) which I did not instruct it to do.

[image]

Am I meant to teach my students to do the math to add one to every loop start and finish because Bipes demands it? This is an unnecessary workaround.

Can you please remove the 'minus one' calculation from lists?

In the Get List Block with LED as the variable, it should just say:

[int(LED)]

JorgeGMarques, 2021-07-24:
Hi!
I'll work on #33 , #34 and #35 right now.

JorgeGMarques, 2021-07-24:
I agree with you, but since it's more a design decision I'll ask @rafaelaroca opinion first.

My view is:  we should follow standards not create new ones, others programming languages like might start arrays at one, but it's a simple principle and both C and MicroPython starts at 0.
Also, it creates more problems than solutions, see `for LED in range (8)`.

ptal78, 2021-07-30:
Thanks.

It's just a pain to tell new students that lists start at one, when the code just looks wrong (as seen below).

[image]

It would be more clear for students to set the loop as "0, 11, 2", and then the list wouldn't need the -1

closed 2021-08-09 by JorgeGMarques