BIPES/BIPES #36: Bug - logic 'if' block failing to add 'else if', or 'else' extensions to the block

ptal78, 2021-07-28:
Doing this:

[image]

Would make you think it will extend the block to add the else if extension, and the else extension to the if block, but it doesn't.

This was tested using the ESP32.

rafaelaroca, 2021-07-28:
Hello. This works perfectly for me. Just tested on ESP32.

[image]

[image]

Could you give more details? It really works correctly to me.

JorgeGMarques, 2021-07-28:
Sometimes this bug happens to me, but I couldn't figure out whats is triggering [Mutators](https://developers.google.com/blockly/guides/create-custom-blocks/extensions) to stop working.

My guess is some `.setTimeout` not resolving and bricking the mutators.

rafaelaroca, 2021-07-28:
Wow. Never happened to me. Curious to know more about it...using Google Chrome?

ptal78, 2021-08-09:
It's working now. Yes, Google Chrome.

What web browser and version do you use for testing?

JorgeGMarques, 2021-08-09:
We use Firefox > 90 and Chrome > 93.

We made some big code refactoring in the last week, unifying the code between `beta2` and `beta2serial` and improving many modules, the platform should be more stable now.
For example, the terminal now works with touchscreens, the Files tab fail way less when fetching, listing and writing files (just still need some styling).

Anyway, if the bug seems to be completely resolved, please close the issue.

closed 2021-08-10 by JorgeGMarques