BIPES/BIPES #44: OLED display blocks for drawing lines

ptal78, 2021-08-19:
Hi,

Can you please add some blocks to use the gfx library for the OLED ssd1306?

This following code works for me once the gfx.py file is loaded

```
from machine import Pin, I2C
import ssd1306
from time import sleep
import gfx

# ESP32 Pin assignment
i2c = I2C(1, scl=Pin(22), sda=Pin(21))

# ESP8266 Pin assignment
#i2c = I2C(-1, scl=Pin(5), sda=Pin(4))

oled_width = 128
oled_height = 64
oled = ssd1306.SSD1306_I2C(oled_width, oled_height, i2c)

graphics = gfx.GFX(oled_width, oled_height, oled.pixel)

while True:

  graphics.line(0, 0, 127, 20, 1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.rect(10, 10, 50, 30, 1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.fill_rect(10, 10, 50, 30, 1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.circle(64, 32, 10, 1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.fill_circle(64, 32, 10, 1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.triangle(10,10,55,20,5,40,1)
  oled.show()
  sleep(1)
  oled.fill(0)

  graphics.fill_triangle(10,10,55,20,5,40,1)
  oled.show()
  sleep(1)
  oled.fill(0)
```

Blocks I suggest to start are:

Line
Rectangle
Triangle
Circle

Rectangle, Triangle, and Circle could each have a filled/outline option

But I'd also like to have blocks to simulate robot motion:
Turtle Turn block and Turtle Move block, although this might be a bit harder to implement.

Phil

ptal78, 2021-08-21:
I've written a small library called tank that works on the ssd1309 to draw lines with move and turn commands:

```
# Header
import tank

# Tank INIT block
tank1 = tank.Tank(64, 32, 90, oled, oled_width, oled_height, i2c)

# Example Code blocks
tank1.turn(45)
tank1.move(30)
```

ptal78, 2021-08-21:
Can you also possibly make the move block with one more variable so it has three states (1=Draw, 2=Store, 3=Move) ?

tank1.move(30, 1)     # pixels, state

Also, a Show OLED block for the OLED would be helpful:

oled.show()

ptal78, 2021-08-21:
Tank Library Code

```
    #  Tank Drawing (Driving practice)
    #  by Philip Tallents

from machine import Pin
from machine import I2C
import ssd1306
import gfx
import math

class Tank:
    def __init__(self, xpos, ypos, direction, oled, width, height, i2c):
        self.xpos = xpos
        self.ypos = ypos
        self.direction = direction
        self.oled = oled
        self.width = width
        self.height = height
        self.i2c = i2c
        self.graphics = gfx.GFX(self.width, self.height, oled.pixel)
        self.state = 1

    def getpos(self):
        return self.xpos, self.ypos, self.direction

    def setpos(self, xpos, ypos, direction):
        self.xpos = xpos
        self.ypos = ypos
        self.direction = direction

    def home(self):
        self.xpos = 0
        self.ypos = 0
        self.direction = 0

    def centre(self):
        self.xpos = self.width/2
        self.ypos = self.height/2

    def move(self, move, state):
        self.xpos = self.xpos % self.width
        self.ypos = self.ypos % self.height
        self.state = state
        oldxpos = self.xpos
        oldypos = self.ypos
        self.xpos = self.xpos + move*math.cos(math.radians(self.direction))
        self.ypos = self.ypos + move*math.sin(math.radians(self.direction))

        if self.state == 0:
            self.state = 0
            
        elif self.state == 1:
            self.graphics.line(int(oldxpos), int(oldypos), int(self.xpos), int(self.ypos), 1)
            self.oled.show()
        elif self.state == 2:
            self.graphics.line(int(oldxpos), int(oldypos), int(self.xpos), int(self.ypos), 1)

    def turn(self, turn):
        self.direction = (self.direction + turn) % 360

    def orient(self, orient):
         self.direction = orient
```

ptal78, 2021-08-21:
gfx library

```
# Port of Adafruit GFX Arduino library to MicroPython.
# Based on: https://github.com/adafruit/Adafruit-GFX-Library
# Author: Tony DiCola (original GFX author Phil Burgess)
# License: MIT License (https://opensource.org/licenses/MIT)

class GFX:

    def __init__(self, width, height, pixel, hline=None, vline=None):
        # Create an instance of the GFX drawing class.  You must pass in the
        # following parameters:
        #  - width = The width of the drawing area in pixels.
        #  - height = The height of the drawing area in pixels.
        #  - pixel = A function to call when a pixel is drawn on the display.
        #            This function should take at least an x and y position
        #            and then any number of optional color or other parameters.
        #  You can also provide the following optional keyword argument to
        #  improve the performance of drawing:
        #  - hline = A function to quickly draw a horizontal line on the displ.
        #            This should take at least an x, y, and width parameter and
        #            any number of optional color or other parameters.
        #  - vline = A function to quickly draw a vertical line on the display.
        #            This should take at least an x, y, and height paraemter a
        #            any number of optional color or other parameters.
        self.width = width
        self.height = height
        self._pixel = pixel
        # Default to slow horizontal & vertical line implementations if no
        # faster versions are provided.
        if hline is None:
            self.hline = self._slow_hline
        else:
            self.hline = hline
        if vline is None:
            self.vline = self._slow_vline
        else:
            self.vline = vline

    def _slow_hline(self, x0, y0, width, *args, **kwargs):
        # Slow implementation of a horizontal line using pixel drawing.
        # This is used as the default horizontal line if no faster override
        # is provided.
        if y0 < 0 or y0 > self.height or x0 < -width or x0 > self.width:
            return
        for i in range(width):
            self._pixel(x0+i, y0, *args, **kwargs)

    def _slow_vline(self, x0, y0, height, *args, **kwargs):
        # Slow implementation of a vertical line using pixel drawing.
        # This is used as the default vertical line if no faster override
        # is provided.
        if y0 < -height or y0 > self.height or x0 < 0 or x0 > self.width:
            return
        for i in range(height):
            self._pixel(x0, y0+i, *args, **kwargs)

    def rect(self, x0, y0, width, height, *args, **kwargs):
        # Rectangle drawing function.  Will draw a single pixel wide rectangle
        # starting in the upper left x0, y0 position and width, height pixels
        # size.
        if y0 < -height or y0 > self.height or x0 < -width or x0 > self.width:
            return
        self.hline(x0, y0, width, *args, **kwargs)
        self.hline(x0, y0+height-1, width, *args, **kwargs)
        self.vline(x0, y0, height, *args, **kwargs)
        self.vline(x0+width-1, y0, height, *args, **kwargs)

    def fill_rect(self, x0, y0, width, height, *args, **kwargs):
[... 137 lines cut ...]
        sb = 0
        if y1 == y2:
            last = y1
        else:
            last = y1-1
        for y in range(y0, last+1):
            a = x0 + sa // dy01
            b = x0 + sb // dy02
            sa += dx01
            sb += dx02
            if a > b:
                a, b = b, a
            self.hline(a, y, b-a+1, *args, **kwargs)
        sa = dx12 * (y - y1)
        sb = dx02 * (y - y0)
        while y <= y2:
            a = x1 + sa // dy12
            b = x0 + sb // dy02
            sa += dx12
            sb += dx02
            if a > b:
                a, b = b, a
            self.hline(a, y, b-a+1, *args, **kwargs)
            y += 1
```

rafaelaroca, 2021-08-21:
Hi Philip! How are you? You have been helping debug and improve BIPES so much, that I feel you are already part of our dev team. In that way, we would like to invite you to be listed on our page as part of our team and help BIPES project more closely.

About your requests, including the libraries is quite easy. I propose that you add them!! (If it is okay for you). 

To add the libraries:
1. Fork BIPES on github (https://github.com/BIPES/BIPES/)
2. On your fork, go to the pylibs folder and add the lib files you created, and others needed, with adequate credits. For example, considering my fork, I would add to: https://github.com/rafaelaroca/BIPES-1/tree/master/ui/pylibs
3. Edit the toolbox.xml and add a button to install the lib:

For example, on the file:
https://github.com/rafaelaroca/BIPES-1/blob/master/ui/toolbox/esp32.xml

Add one or two lines to show the install lib button, such as:
      <label text="Library: https://github.com/devbis/st7789py_mpy"></label>
      <button text="Install st7789py library" callbackKey="installPyLib"></button>

After making these updates, perform a git commit and git push. This will update your BIPES repository. Then create a "pull request", and we will merge on the main repo and server to everyone.

If this works for you, you can add several other libs!

Now, after adding the library and the button to install the library, we have to create the blocks for the library.

One quick and easy possibility, is using blockly block factory:

https://blockly-demo.appspot.com/static/demos/blockfactory/index.html

At the link above, you can "draw" the blocks, most possible options. The tool will give you 2 pieces of files, that you can copy and paste the created lines to the end of code/generator_stubs.js and core/block_definitions.js. Then, just add the block to the toolbox XML file! Finally, edit the generator_stubs.js to add the generated code for that block.

Finally, I have written this example/instructions to create new blocks (thanks to you!). It may be helpful:

https://bipes.net.br/wp/creating-new-blocks/

What do you think? Worth a try? 

We could add the libs and create the blocks for you, but maybe you would like to do it yourself.

best regards
Rafael
Rafael

rafaelaroca, 2021-08-21:
Just a quick comment: some function calls you are asking for are already on the library we are using:

http://bipes.net.br/beta2/ui/pylibs/st7789py.py

rect, line, fill, vline, hline are already available. I just did not have the time/priority to create the blocks for them :-(

ptal78, 2021-08-21:
And one added feature, can the Init block for the Tank block have a name option? This would allow the user to make multiple tank objects.

And each tank block would need a drop-down selector to choose from the created tank names.

ptal78, 2021-08-22:
Hi Rafaelaroca,

Thanks for the invite to join developing Bipes. I'm happy to be part of your team in the spare time I have. It's fantastic software.
It's going to take me quite some time to figure out the Block Factory and everything.

Are you possibly able to just show me a basic implementation of the three tools, Tank Init, Tank Move, and Tank Turn?

ptal78, 2021-08-22:
So far I have started creating one block called tank_move - and here's the library for the three blocks, but I'm just not sure about Java code yet. Can you possibly help?

```
<xml xmlns="https://developers.google.com/blockly/xml"><block type="factory_base" id="5MWyc~F}DSGC5,d)eOK{" deletable="false" movable="false" x="10" y="10">
    <mutation connections="BOTH"></mutation>
    <field name="NAME">tank_move</field>
    <field name="INLINE">INT</field>
    <field name="CONNECTIONS">BOTH</field>
    <statement name="INPUTS">
      <block type="input_dummy" id="CL!2w|/Bi:=^K)q$)([,">
        <field name="ALIGN">LEFT</field>
        <statement name="FIELDS">
          <block type="field_static" id="g|f5PG`jx.Qc-Q.v0:|]">
            <field name="TEXT">Tank to</field>
            <next>
              <block type="field_dropdown" id=";1[E+g#^[p;86iV#6c33">
                <mutation options="[&quot;text&quot;,&quot;text&quot;,&quot;text&quot;]"></mutation>
                <field name="FIELDNAME">option</field>
                <field name="USER0">Move</field>
                <field name="CPU0">TankMove</field>
                <field name="USER1">Prime</field>
                <field name="CPU1">TankPrime</field>
                <field name="USER2">Jump</field>
                <field name="CPU2">TankJump</field>
              </block>
            </next>
          </block>
        </statement>
        <next>
          <block type="input_value" id="DLk-eTgzz]aKZe@IX[W-">
            <field name="INPUTNAME">Move</field>
            <field name="ALIGN">LEFT</field>
            <statement name="FIELDS">
              <block type="field_static" id="]X=b30:6pgs,BL0LeCgQ">
                <field name="TEXT">distance of</field>
              </block>
            </statement>
            <value name="TYPE">
              <shadow type="type_null" id="[Hv]J_jo-IM[NIcRb73h"></shadow>
              <block type="type_number" id="wuP2)J}g?mY|k(16H/{p"></block>
            </value>
          </block>
        </next>
      </block>
    </statement>
    <value name="TOOLTIP">
      <block type="text" id="t}e1^f(g8c]sjW$ShOq{" deletable="false" movable="false">
        <field name="TEXT">Move, Prime, or Jump the Tank a 'pixel' distance</field>
      </block>
    </value>
    <value name="HELPURL">
      <block type="text" id="+@h-uzAXaIc?Ald8%0ZU" deletable="false" movable="false">
        <field name="TEXT"></field>
      </block>
    </value>
    <value name="TOPTYPE">
      <shadow type="type_null" id="/%(Hz_y`LQPUJMXtvV}M"></shadow>
    </value>
    <value name="BOTTOMTYPE">
      <shadow type="type_null" id="BVTPFSsbldJ].NLfkDn2"></shadow>
    </value>
    <value name="COLOUR">
      <block type="colour_hue" id="f6DT@8E=rWxdJ|zY#5br">
        <mutation colour="#805ba5"></mutation>
        <field name="HUE">270</field>
      </block>
    </value>
  </block><block type="factory_base" id="3p4Ek~JLNzisYnx{|q2z" deletable="false" movable="false" x="10" y="10">
    <mutation connections="BOTH"></mutation>
    <field name="NAME">tank_turn</field>
[... 104 lines cut ...]
    </statement>
    <value name="TOOLTIP">
      <block type="text" id=";/doyI)`Mql8(zrcV7PC" deletable="false" movable="false">
        <field name="TEXT">Tank Init creates a positional 'cursor' object to help you track motion at a desired length and angle.</field>
      </block>
    </value>
    <value name="HELPURL">
      <block type="text" id="81P#(@8-s}S+:hpiKv:Q" deletable="false" movable="false">
        <field name="TEXT"></field>
      </block>
    </value>
    <value name="TOPTYPE">
      <shadow type="type_null" id="w@#Z]b~AS:b46-NksM?o"></shadow>
    </value>
    <value name="BOTTOMTYPE">
      <shadow type="type_null" id="4BOSs+iyxZ/gPyE;Ljyj"></shadow>
    </value>
    <value name="COLOUR">
      <block type="colour_hue" id="6LAbd;=LxEc?GFa#@wu3">
        <mutation colour="#805ba5"></mutation>
        <field name="HUE">270</field>
      </block>
    </value>
  </block></xml>
```

ptal78, 2021-08-22:
The xaml code above will load the blocks I've created in the block factory, but I'm not sure how to link the blocks to generate the code.

ptal78, 2021-08-22:
I'm doing it now. Doesn't seem so hard now that I've found the stub and js files.

ptal78, 2021-08-22:
Hi Rafaelaroca,

I have modified the js generator and stubs and xml for the files, but being new to github, I'm not sure I have 'pushed' the files correctly before I did the 'pull request'. I have modified some more and don't know if you have these.

Can you see the necessary file changes?

Also, I can't test the files out yet, so are you able to test if I've done it right? 
It's getting late here.

Kind Regards,
Phil

ptal78, 2021-08-22:
I got make to run, but make offline had errors.

The error said: 

bipes_offline.zip *
/bin/sh: 1: zip: not found
Makefile:21: recipe for target 'offline' failed
make: *** [offline] Error 127

ptal78, 2021-08-22:
I didn't have zip installed.

It now runs make offline successfully. But I checked the newly created zip files in windows and the changes screwed up the OLED display toolbox. I'm looking into it now. Sorry for the trouble this may have caused.

JorgeGMarques, 2021-08-22:
Hey phil,

don't worry, fork changes don't affect the master branch until merged.
I reviewed your code at: https://github.com/BIPES/BIPES/pull/49

About the `index_offline.html`. we don't use the bot anymore because it commits the whole platform inside a zip and huge html file.
Also data generated by `make offline` is not meant to be committed, we even [git ignore](https://git-scm.com/docs/gitignore) `.zip` and  `index_offline.html` to avoid commiting it.
Therefore, I need you to disable in your fork the bot at https://github.com/ptal78/BIPES_ui_testing/actions ? Just click the 3 dots and choose "disable workflow".
[image]

ahunaxi, 2024-09-18:
@ptal78 did you manage to make the tank blocks to work? I am also working with the SSD1306 on the ESP32 and the Tank Init blocks don't seem to render anything; even though I run the Update Display right after it.