BIPES/BIPES #45: Hard to debug now the Code tab is gone

ptal78, 2021-08-20:
Are for debug of code is needed to understand why blocks aren't working without loading the code on the device.

Eg. why is the Init ST7789 block not working? I'd understand better if I could see the micropython code being used.

JorgeGMarques, 2021-08-20:
It's here actually!
[image]

Just leave this "file" open and it'll work exactly like the "Code" tab (automatically refreshes when changing the blocks).

I moved because now you can edit/debug BIPES XML file, and theoretically we could do both blocks -> micropython as micropython -> blocks

ptal78, 2021-08-21:
Cheers

closed 2021-08-21 by ptal78