BIPES/BIPES #46: Init ST7789 block is not working

ptal78, 2021-08-20:
I wrote the code to my device and it seems the Init ST7789 block is not connected to block yet.

eg. The block says I2C with SCL=5, and SCD=4, but the code generated (as it should be) is generated for SPI protocol.
Please update the Init ST7789 block so it represents the SPI connections with SCK, MOSI, Reset, CS, DC.

```
=== import machine
=== import st7789py
=== 
=== 
=== # turn on backlight
=== bl = machine.Pin(4, machine.Pin.OUT)
=== bl.value(1)
=== spi = machine.SPI(1, baudrate=20000000, polarity=1, phase=1, sck=machine.Pin(18), mosi=machine.Pin(19))
=== display7789 = st7789py.ST7789(spi, 135, 240, reset=machine.Pin(23, machine.Pin.OUT), cs=machine.Pin(5, machine.Pin.OUT), dc=machine.Pin(16, machine.Pin.OUT))
=== display7789.init()
=== display7789.fill(st7789py.color565(40, 40, 1))
```

rafaelaroca, 2021-08-29:
Fixed.

rafaelaroca, 2021-08-29:
By the way, a working example: http://bipes.net.br/beta2/ui/#fvquhu

closed 2021-08-29 by rafaelaroca