BIPES/BIPES #48: WebBluetooth files not found - library code error

ptal78, 2021-08-21:
I just tried to load the library file for WebBluetooth, and the three files downloaded succesfully onto my ESP32 contained the following code:

```
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL was not found on this server.</p>
<hr>
<address>Apache/2.4.38 (Debian) Server at bipes.net.br Port 80</address>
</body></html>
```

ptal78, 2021-08-21:
The files also save on the esp32 with the extension .py.py

ptal78, 2021-08-21:
I found the correct files on github micropython/examples/bluetooth/

These seemed to work well without a problem. But I suggest it might work well if you can somehow get some setup dialog or block to edit the name and UUID options of the bluetooth device ble_uart_peripheral.py library file before it loads onto the ESP32.

```
_UART_UUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX = (
    bluetooth.UUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E"),
    _FLAG_NOTIFY,
)
_UART_RX = (
    bluetooth.UUID("6E400002-B5A3-F393-E0A9-E50E24DCCA9E"),
    _FLAG_WRITE,
)
_UART_SERVICE = (
    _UART_UUID,
    (_UART_TX, _UART_RX),
)

# org.bluetooth.characteristic.gap.appearance.xml
_ADV_APPEARANCE_GENERIC_COMPUTER = const(128)

class BLEUART:
    def __init__(self, ble, name="mpy-uart", rxbuf=100):
```

rafaelaroca, 2021-08-21:
Hello.Fixed!

You may have to reload BIPES several times due to cache. 

You can check if the correct version is loaded when the toolbox shows "Install ..... file" (without .py).

regards
Rafael

closed 2021-08-21 by rafaelaroca

rafaelaroca, 2021-08-21:
> I found the correct files on github micropython/examples/bluetooth/
>
> These seemed to work well without a problem. But I suggest it might work well if you can somehow get some setup dialog or block [...]

Have also added the option of defining the BLE name, as you asked. Reload it to use :-)