BIPES/BIPES #59: Offline Bipes doesn't seem to upload files

ptal78, 2021-08-31:
Upload works great while online at http or https, but the offline version doesn't seem to upload anything. Has this been updated with the new upload code?

ptal78, 2021-08-31:
Or is there possibly a broken link in the offline version?

rafaelaroca, 2021-08-31:
Any error on the javascript console or BIPES screen / notifications?

JorgeGMarques, 2021-09-05:
Yes, you can't upload while running locally due to the [CORS policy](https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSRequestNotHttp).
It's the same reason for `index_offline.html` , but in this case we were able to just concatenate everything we need inside this HTML file.
Also, we can't access [iframes](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/iframe) , so [Freeboard](https://freeboard.io/) projects won't be included in the BIPES XML file when running locally.

There is nothing I can do about it, sorry...

closed 2021-09-05 by JorgeGMarques