BIPES/BIPES #60: PWM deinit tool

ptal78, 2021-09-05:
Hi,

It would be great if you could implement a PWM deinit block that allows the user to select from the existing PWM options.

I'm not sure how to do this in code to reference other existing blocks.

But the code is simple.

Eg.

pwm4.deinit()

Of course the user would someone need to select the pwm pin to deinitialize.

I don't have time to work on this though sorry.

Cheers

JorgeGMarques, 2021-09-05:
Included with e59445c9948da88042a4d316a784d7e75523827d.

closed 2021-09-05 by JorgeGMarques

ptal78, 2021-09-06:
Thanks. But it's not working yet because the number has been put in brackets as shown below, when it should be without brackets ie, pwm5.deinit() not pwm(5).deinit()

[image]

ptal78, 2021-09-06:
Also, as you can see from the code example, the gpio_set commands actually stop the pwm5 from working, as they redefine the pin. So it is quite important that the pwm5 = PWM(Pin(5)) line should be placed directly prior the pwm5.freq line.