BIPES/BIPES #63: Offline Bipes missing new tools such as the tank tool in the OLED Displays

ptal78, 2021-09-09:
Hi, we really need the tank tool in the offline version of Bipes to use next week if possible.

Currently, the tank tools, and any new tools are not being added into the offline version like they used to be.

JorgeGMarques, 2021-09-09:
Get the latest release at https://github.com/BIPES/BIPES/releases/tag/v0.2.2-beta
It's under  `Displays -> OLED Display` for the ESP32.

ptal78, 2021-09-14:
Hi Jorge,

The offline Biped still doesn't have the new own tools.

Can you possibly add these in ASAP?

Kind regards,
Phil

ptal78, 2021-09-14:
New PWM tools needed please in Bipes Offline.

Phil

JorgeGMarques, 2021-09-14:
Hi, just made a realease with latest: https://github.com/BIPES/BIPES/releases/tag/v02.2.2.1-beta

closed 2021-09-15 by JorgeGMarques

ptal78, 2021-09-15:
Thanks Jorge,

But the zip file doesn't seem to have the offline html document anymore.

Is there a reason this is no longer in the ui folder?

Regards,

*Philip Tallents*

*Manager*
*PicoKit*
ABN: 88 995 830 437
Ph: 0402 239 363
Ph: (07) 5330 3095
Fax: (07) 3465 0154

ptal78, 2021-09-15:
Found it.

https://github.com/BIPES/BIPES/releases/download/v0.2.2.1-beta/bipes_offline.zip

Regards,

*Philip Tallents*

*Manager*
*PicoKit*
ABN: 88 995 830 437
Ph: 0402 239 363
Ph: (07) 5330 3095
Fax: (07) 3465 0154