BIPES/BIPES #79: How to set correct blockly for Linux system?

WolfOptics, 2021-12-03:
Hi sirs,
I got some problem using blockly in linux system as shown as picture.
https://drive.google.com/file/d/1n4EhUH57qYAP2y8ppRB7OT9e7SsM2ooh/view?usp=sharing

Could you help me? It was NOT workable in my raspberry pi 3B+.

Best Regards,

JorgeGMarques, 2021-12-03:
@rafaelaroca Can you provide some insight on how to expose the python shell to BIPES?

WolfOptics, 2021-12-04:
Hi sir,
 The information was shown as below in my Linux's terminal after I push "Run block based program" in BIPES.

[https://drive.google.com/file/d/1FrJYa0WBIC-BUK8VhASxTx0uSY1r5ynG/view?usp=sharing](url)

Best Regards,

rafaelaroca, 2021-12-18:
Hello @WolfOptics, sorry for the delay.

Did you try to run SocketServerWebREPL on the board?

Like this:

First start SocketServerWebREPL:

$ git clone https://github.com/BIPES/SocketServerWebREPL
$ cd SocketServerWebREPL/
$ python2 socketserverWebREPL.py 
BIPES Project - Bridge started
Now, access the Python Console from a Web Browser using a WebREPL client
For example: https://micropython.org/webrepl/#127.0.0.1:8266/

You can run the commands above on your Raspberry Pi or any other Linux device. Even your PC, for testing. After SocketServerWebREPL is started, got to BIPES and fill the IP address:

[image]

Click on the "Network" text, which means "Connect":

You should see this message at the terminal:

[image]

Type s and you should get the Python REPL:

[image]

Then you can try some commands:

[image]

Now, seems that there is a bug for sending block based code to the host. We will check it and get back.

Hope this helps for a first test.

Best regards.

andrewg78, 2021-12-25:
I can confirm the issue
[image]
.

rafaelaroca, 2021-12-25:
Hi @andrewg78, thanks for testing. Nice to know. Did the console output "ABC", or the program crashed? Thanks for helping debug BIPES!

andrewg78, 2021-12-25:
1. I can send blocks several times
 through the 'Run block based program' It hits rasbperry PI every time, but there is a new line char after the 'b' [image]
[image]

2. When I refresh BIPES site (F5), SocketServerWebREPL receives last char from the BIPES console in the infinite loop!!!  [image]
3. Python Remote Shell sends python code properly No issue here!!!
4. After I send block through 'Run block based program'  prompt stop responding on  ' Python Remote Shell ' ( I do not see what I'm typing)

rafaelaroca, 2021-12-26:
Hello @andrewg78, just updated https://github.com/BIPES/SocketServerWebREPL. Could you please update your SocketServerWebREPL and try again? Running programs is now reliable, but there are still other bugs. But one thing at a time :-)

rafaelaroca, 2021-12-26:
> Hello @andrewg78, just updated https://github.com/BIPES/SocketServerWebREPL. Could you please update your SocketServerWebREPL and try again? Running programs is now reliable, but there are still other bugs. But one thing at a time :-)

By the way, here is a working example:

Link to the program: 
http://bipes.net.br/beta2/ui/#9ec4ro

Program:
[image]

Result:
[image]

andrewg78, 2021-12-27:
Hi. I see some progress, although there are still some glitches ;)
So:
1. What is the expected outcome on the Python Remote Shell and on the LInux terminal? I see two different results. Should we have the same outputs ?
2. There is a syntax Error at the end.
[image]

3. When Python Shell is not started(s key not pressed), but block based program is run, we got this
[image]

4. when I start socket server as a non sudo user, program is executed from blocks but we got a lot of 'IndentationError: unexpected indent' 
[image]

5. When I run with sudo (sudo python socketserverWebREPL.py) there is NO issue with 'unexpected indent'
6. When I use mouse scroll in the Remote Python Shell, BIPES disconnects from the  socket server and we  must again click on Connect to restore the connection
[image]
7. System commands output always '0' 
[image]

rafaelaroca, 2021-12-27:
> Hi. I see some progress, although there are still some glitches ;) So:
>
> 1. What is the expected outcome on the Python Remote Shell and on the LInux terminal? I see two different results. Should we [...]

Hi @andrewg78, we are advancing. Thank you for helping!

1. Run Linux Command is expected to run a command on the remote board, but without getting the output. I use it, for example, to call mpg123 and start playing mp3 music, or to call speak so that the remote device can speak text from a BIPES command. So, the outputs are different, because the command output is shown on the remote terminal, but we could change it so that it is also shown on the BIPES console.

2. Sorry, this is a bug. Tested and confirmed. We will fix it, but not that that code execution works. This error may happen due to some garbage at the end of the buffer.

3. This is ok. Without Python shell started, no Python statement should work. No commands should be sent before starting Python session (s sent).

4 and 5. This issue did not occur to me for both normal execution and execution with sudo. Maybe your default python interpreter is different between these users (user default user and root)?

6. Will check it later. I did not test/replicate the issue.

7. Ok, the command specified on os.system() runs, as you showed on the remote terminal. What happens is that the STDOUT and STDERR of the executed command is not redirected to BIPES console. Actually, we did not implement this, so we can have it on our roadmap. Anyway, if you need the program output you can build a block for that!! :-)
with the command: x = os.popen('date').read()

For example:

[image]

The above example creates a function that receives an argument and execute the argument as a Linux command, getting the result and returning to the main program to be used.

rafaelaroca, 2021-12-27:
Sorry, did not test the last example. This one is tested / working:

[image]

"Run Linux Command and get output" was created using the "Functions" item of the toolbox. The function receives a command as argument, builds the command with os.popen(command).read() and evaluates the result to the out variable, which is returned. Note the test with several commands.

Direct link to the program:
http://bipes.net.br/beta2/ui/#2azfub

rafaelaroca, 2021-12-28:
> Hi. I see some progress, although there are still some glitches ;) So:
>
> 1. What is the expected outcome on the Python Remote Shell and on the LInux terminal? I see two different results. Should we [...]

Hello. Item 2 (syntax error at the end) is now fixed. Please, get the latest version on github.

[image]

andrewg78, 2021-12-28:
Hi. Confirmed:
issue number 2 is fixed! Good job!
issues 4 and 5 also fixed ( sudo not needed on the newest version)
Regarding the issue no3 .I would argue here ;)
This is not user friendly. For instance, in order to get the file list I must first go to Console Tab  and Enable Remote Shell with the 's' key.
Can we have Python Shell started automatically after the connection is established ?

andrewg78, 2021-12-28:
Issue no 6 - I updated description in the original post. This one should be closed as well.

rafaelaroca, 2021-12-28:
Hi @andrewg78 !

Agree with your point on issue number 3. It is not friendly, however, a minimal authentication would be nice. In that way, I have just updated https://github.com/BIPES/SocketServerWebREPL so that SocketServerWebREPL asks for a password when the connection is established. By default, the password is "bipes". You can have the password typed on the connection dialog box, so it will be automatically sent, and the connection will be established directly, as you suggested. 

Password should be typed here (and will be saved):
[image]

Well, finally, now the shell is started automatically, as you suggested!

andrewg78, 2021-12-29:
Issue no 3 is resolved. Gr8!!!
I think we should move some of your responses and code examples to WIKI or to documentation page. Do you have one?  I can copy-paste it to wiki right away. Just let me know.
There is still minor issue with websocket server when BIPES page is refreshed, but I will create separate issue on that. This thread can be closed I think

rafaelaroca, 2021-12-29:
Hi @andrewg78. Great! Closing this issue! Thank you very much for helping debug it!

closed 2021-12-29 by rafaelaroca