BIPES/BIPES #83: beta2

andrewg78, 2021-12-25:
beta2 is hardcoded in several requests
It leads to 404 when BIPES is used in local webserver
[image]

rafaelaroca, 2021-12-25:
You are right. Sorry for that and thanks for reporting. We will work on making these paths relative.

andrewg78, 2021-12-25:
I can rise a pull request on that. I did fix this on my end already.
Not sure about urls to https://bipes.net.br/beta2/ui/
I did not change them, but maybe they should be  relative as well?
[image]

JorgeGMarques, 2021-12-25:
Yes, every url should be relative! We even have been using another url for a while now (bipes.net.br/**ide**  instead of bipes.net.br/**beta2/ui**).

Feel free to open a pull request :)