BIPES/BIPES #87: DHT11 Issue on esp32 in new version

WolfOptics, 2021-12-26:
Hi sirs,
 In the new version environment, the I got issue using DHT11 sensor as below.
How can we solve this issue?
[image]

rafaelaroca, 2021-12-26:
Hello @WolfOptics :-)

This seems to be an issue with the sensor or with the connection. Just tried with an ESP8266 and worked on first try:

[image]

If I disconnect the DATA pin, I get the same error you got: ETIMEDOUT

Hope it helps.

rafaelaroca, 2021-12-26:
Just tried with an ESP32 and it also worked nice! :-)

[image]

So, please, check your sensor and connections.

closed 2021-12-28 by rafaelaroca