BIPES/BIPES #88: Missing Blocks in the Any Linux Mode

andrewg78, 2021-12-28:
There are some blocks missing for Raspberry Pi
For instance, there is no timing, and many more.
ESP32
[image]
RPI
[image]

rafaelaroca, 2021-12-28:
Hi @andrewg78. Sorry, these blocks are missing due to lack of time to add them to the toolbox. Note that you can use blocks from any device! You can select ESP8266 and use the timing block, for example. Adding them to the Linux toolbox is easy, but still did not add them because I did not test these blocks on Linux/Python. Anyway, if you could help us define a list or add these blocks, it would be nice. Addind these blocks to Linux is quite easy: edit ui/toolbox/linux.xml and add the desired blocks and sections. Use ui/toolbox/esp8266.xml as an example, if you want.

andrewg78, 2021-12-29:
OK. I will try.

rafaelaroca, 2024-11-10:
Hello, closing as this is not an issue, but an improvemnt. Hope someone (or me) gets time someday to finish this.

closed 2024-11-10 by rafaelaroca