BIPES/BIPES #96: Cron

andrewg78, 2022-01-06:
When the Websocket server is started from the Linux Cron, the file list can not be retrieved from the device and the connection is closed.
@reboot python /home/pi/SocketServerWebREPL/socketserverWebREPL.py >/home/pi/bipes.log 2>&1
[image]

andrewg78, 2022-01-07:
Workaround:
It looks like  websocket server works (at the moment) as a terminal application only, in order to push it to the background, we can use a screen command, so we can have it autostarted on the boot process:
@reboot screen -dm bash -c "python /home/pi/SocketServerWebREPL/socketserverWebREPL.py"

rafaelaroca, 2022-01-15:
Hi @andrewg78. For now, I will close this issue, as you proposed a great workaround. As SocketServerWrebREPL needs a valid environment, it really needs to be started from the screen or other tool that provides a complete environment. Anyway, running it from screen seems a great solution for me.

closed 2022-01-15 by rafaelaroca