BIPES/BIPES #123: Save and share databaord using IDs

TripleLardon, 2022-05-04:
Hello ! 

I installed my own BIPES server. I installed also databoard and freeboard repos. 
All works fine but I also want to be able to save and share the databoard like on the BIPES official website. 
Is it possible with the basic configuration ? or I need to install additional software, or develop this myself ? 

Thanks for reading :)

gastmaier, 2022-05-05:
Hi, sorry for not replying sooner

The second version uses **Google's App Engine to store the projects in the cloud**, however I wasn't responsible for this and I only know that it comes from [Google's suggested implementation](https://developers.google.com/blockly/guides/configure/web/cloud-storage)... Our App Engine source files are at [appengine/bipes-271213](https://github.com/BIPES/BIPES/tree/master/appengine/bipes-271213).

We are actively working in the new version at the [third branch](https://github.com/BIPES/BIPES/tree/third), which automatically set-ups itself, including project sharing:

```
git clone  --branch third https://github.com/BIPES/BIPES.git
cd BIPES
make
```

You can try it online at [bipes.net.br/3](https://bipes.net.br/3)with documentation at [bipes.net.br/3/docs](https://bipes.net.br/3/docs)

**But we are dropping freeboard support for now** to focus on databoard (renamed to dashboard and included in the main repo).
(The third version do support custom pages/plugins by just cloning additional repositories, however we haven't had the time yet to take advantage of this functionality)

TripleLardon, 2022-05-06:
Hi ! No problem :)

Thanks for your explainations. I'm still trying to make freeboard run. On the folder *easymqtt* I see that on thePHP scripts you require vendor/ libraries. But on the repos it's missing the composer.json file. How can I know which libraries install ? 

Thanks for your good responses !

gastmaier, 2022-05-07:
Hi, I just checked the server.

the compser.json is:
```
{
    "require": {
        "mongodb/mongodb": "^1.7"
    }
}
```
From @caioasilva's [undergraduate thesis](https://repositorio.ufscar.br/handle/ufscar/13656) (in Brazilian Portuguese) I got:

> Inside the folder */var/www/html/easymqtt*, the following were executed commands to configure the MongoDB library:
```
php -r ​"copy('https://getcomposer.org/installer', 'composer-setup.php');"
php -r ​"if (hash_file('sha384', 'composer-setup.php') ===
'795f976fe0ebd8b75f26a6dd68f78fd3453ce79f32ecb33e7fd087d39bfeb978342fb73ac986cd4f
54edd0dc902601dc') { echo 'Installer verified'; } else { echo 'Installer
corrupt'; unlink('composer-setup.php'); } echo PHP_EOL;"
php composer-setup.php
php -r ​"unlink('composer-setup.php');"
php composer.phar require mongodb/mongodb
```

rafaelaroca, 2022-07-23:
Closed as answered by @gastmaier

closed 2022-07-23 by rafaelaroca