BIPES/BIPES #128: Outdated Makefile "copy"-entry?

casjengu, 2022-06-07:
Hello, 

when I run the "make copy" command of the makefile, the storage.js used by BIPES is overwritten by the one used by blockly. This causes problems with multi-project support.

Is the make-copy entry outdated?

Here is an excerpt of the program code:
[image]

gastmaier, 2022-06-08:
It is outdated, but you don't have to do `make copy` in the master branch (second version).

There is a new version of BIPES in the third branch, which includes an "out-of-the-box" back-end to allow project sharing and MQTT. For more information about that see [bipes.net.br/3/docs/contribute](http://bipes.net.br/3/docs/contribute/forking.html).

I'll take a look into merging the third branch into master to avoid walking in circles with this second version and motivate new contributors into the new version instead of the second...

casjengu, 2022-06-09:
Hello Jorge,
thank you for the quick reply and clarification. 
I will definitely take a look at the new BIPES version.