BIPES/BIPES #130: Add a download button

tknobi, 2022-06-09:
Dear BIPES Team,

first of all thank you for your awesome project! :partying_face: We are currently building the [CoRoSoN-Kit](https://github.com/CoRoSoN-Kit/CoRoSoN-Kit), a robotics kit for young students to help them participate in RoboCup Junior Soccer. For this, visual programming would be very helpful, so I recently analyzed several options for this and yours seems by far the most mature. (See our [Issue 28](https://github.com/CoRoSoN-Kit/CoRoSoN-Kit/issues/28))  :slightly_smiling_face: 

The only disadvantage I can see is that there is no download button. You can start the program in the upper right corner, but after restarting the processor it does not start on its own. I know about the manual process of copying the created textual program. This solution is probably a big hurdle for beginners. 

Would it be possible to build a "download as autostart program" button next to the run button? Or could the run button also save the program permanently as an "autostart program"? 

Thank you very much
tknobi

rafaelaroca, 2022-06-10:
Hello @tknobi 

Thank you for letting us know about what you are doing with BIPES! We are very happy with that. By the way, we have BIPES Team members involved with the RoboCup organization! 

About the "download as autostart program", maybe there is already an easy way!

You can simply click on Files -> type main.py on the file name and click save! It is not necessary to copy & paste (in the image below, click on my_project.bipes.py and type main.py and then save)

[image]

If this does not solve the issue, let us know to discuss other possibilities. Your suggestion would be a nice addition, anyway

best regards.

gastmaier, 2022-06-10:
Beyond @rafaelaroca suggestion, I have some questions.

Would you like to integrate the board to (a custom version of) BIPES?
Are you more interested in using the [third version](https://bipes.net.br/3) (in branch third) or the [second version](https://bipes.net.br/ide) (in master)?

Best regards

tknobi, 2022-06-13:
Hey @rafaelaroca & @gastmaier ,

thank you for the kind words and nice to hear that you are also involved in RoboCup. (Together we will get it done, that the robots will win against the FIFA world champions in 2050 :soccer: :robot: :laughing:  ).

> You can simply click on Files -> type main.py on the file name and click save! It is not necessary to copy & paste (in the image below, click on my_project.bipes.py and type main.py and then save)

I have tested your suggestion. Yes, it simplifies the copy & paste, however I still find it quite cumbersome and too complicated for newcomers. The named procedure must be executed after each program change.

May I kindly ask why it is not written to the main.py file by default? I understand that you want to have multiple programs on the processor, but isn't the most common use case "1 board -> 1 program" and should run by autostart after download? To be able to flash multiple programs could still be possible after explicit switching.

> Would you like to integrate the board to (a custom version of) BIPES?

Even though BIPES is my favorite, we haven't made a final decision as a team. But yes, if we use BIPES we will need our own blocks. For example an IR receiver sensor block or an omniwheel driving block. 
I would then like to integrate these blocks into BIPES. We are happy to send them back to your official repo via pull requests, if you want.

> Are you more interested in using the [third version](https://bipes.net.br/3) (in branch third) or the [second version](https://bipes.net.br/ide) (in master)?

To be honest, I didn't know about version 3 until your comment, is it already stable? Which version would you recommend?

Best regards
tknobi