BIPES/BIPES #132: Settings are not persistent / saved

wohltat, 2022-06-19:
Settings like the device and the ip address, password, etc. are not persistent. 
It also does not end up in the project file.

Pretty inconvenient to enter this every time. Some settings are already gone on a page refresh. 
I don't know if i'm doing something wrong. In the old version of bipes it worked fine.

gastmaier, 2022-06-19:
By new version, do you mean the third (https://bipes.net.br/3/) or the second (https://bipes.net.br/ide/)?

I have some free time later today, I can do it for the third version, I was already planning into fixing some bugs.

wohltat, 2022-06-19:
Sorry, forgot to mention.
I mean the online version https://bipes.net.br/ide/ . So the second version.
Didn't even know about the third version, because it was not linked on the main page.

The data seems not be persistent in the version 3 as well.
Furthermore i cannot choose the esp8266 (wemos) boards there.

Is the online version the latest or does it make sense to use self host the latest version from git?

gastmaier, 2022-06-19:
The online version is the latest for both versions (second and third)
But I do like when people self-host ([docs on how to self-host the third version](https://bipes.net.br/3/docs/contribute/forking.html)).

I'll look into enabling back all boards and making the settings persistent for the third version later today.

closed 2022-06-19 by gastmaier

wohltat, 2022-06-20:
Is version 2 not changed anymore?

It would be nice to also have the fix there since version 3 is not quite where version 2 is in terms of features and stability.

gastmaier, 2022-06-21:
Well, it would be double the work for me, and version 2 have some awful core issues that have already been fixed in the third version.