BIPES/BIPES #133: shared files not accessible

wohltat, 2022-06-19:
If i click on 'Click here to view shared programs' nothing happens
It is linked to https://bipes-271213.appspot.com/list

gastmaier, 2022-06-19:
If you wait a lot, it eventually loads... It is just terrible inefficient because it uses Google's app engine.
The third version uses PostgreSQL directly and is much faster and is embedded at https://bipes.net.br/3/ide?page=project but has little to none projects shared yet (to share projects, right ciick a project in "Your projects" list).

wohltat, 2022-06-19:
You are right. After some minutes the list showed up.

I like the version 3. But some features seems to be missing and it is still less stable.

closed 2022-06-19 by gastmaier