BIPES/BIPES #138: MQTT connect uses wrong parameter type

wohltat, 2022-06-19:
the EasyMQTT connect does not work. The port is given as a string which causes the call to fail.

```>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== easymqtt_session = "qi8f27";
=== easymqtt_client = umqtt.robust.MQTTClient("umqtt_client", server = "bipes.net.br", port = "1883", user = "bipes", password = "m8YLUr5uW3T");
=== easymqtt_client.connect()
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "umqtt/simple.py", line 67, in connect
TypeError: can't convert str to int
>>> easymqtt_session = "qi8f27"
>>> easymqtt_client = umqtt.robust.MQTTClient("umqtt_client", server = "bipes.net.br", port = 1883, user = "bipes", password = "m8YLUr5uW3T")
>>> easymqtt_client.connect()
0
>>> easymqtt_client.publish(easymqtt_session + "/" + 'topik', str(823))
>>> 
```

If the port is an int like in the version 2 of bipes it works.
Either this is a typo or a difference between esp32 and esp8266.

Btw, is the dashboard supposed to work? I picked the matching topic name but nothing changes.

gastmaier, 2022-06-19:
About the dashboard, the MQTT widgets are offline because the encrypted (wss) seemed to be conflicting with the non-encrypted (http) port.
We want to keep the non-encrypted port to keep supporting devices that don't support TLS well.

To check if the MQTT server is online, see if the MQTT session is **not** blinking:

[image]

I hope I can get time soon to fix/finish these.

PS: Thanks for pointing out the issues.

wohltat, 2022-09-19:
any progress? The session session is still blinking.

if i turn the port string into an int the error message changes to
``` python
...
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "umqtt/simple.py", line 67, in connect
TypeError: can't convert str to int
>>> easymqtt_client = umqtt.robust.MQTTClient("umqtt_client", server = "bipes.net.br", port = 1883, user = "bipes", password = "m8YLUr5uW3T");
>>> easymqtt_client.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 67, in connect
OSError: -2
>>> 
```

wohltat, 2022-09-20:
> About the dashboard, the MQTT widgets are offline because the encrypted (wss) seemed to be conflicting with the non-encrypted (http) port.
> We want to keep the non-encrypted port to keep supporting devices that don't support TLS well.

I don't get why this is a problem. The easyMQTT in BIPES IDE version 2 works well and it also is a https web site.

I can send data using `mosquitto_pub` and it appears there:
```
mosquitto_pub -h 34.95.149.23 -t "iysj9n/data/" -m 32
```
[image]

----------------

The browser side is one thing but the micropython side doesn't work in the first place.

The `OSError: -2` from my last post was just because i forgot to set up the network correctly. So mqqt not working on the esp _was basically just the type issue_.

After setting up the network i can connect to the MQTT server and publish some data that is displayed in another tab that runs the bipes v2. So in general this should be possible.

``` python
>>> import umqtt.robust
>>> import time
>>> import utime
>>> easymqtt_client = umqtt.robust.MQTTClient("umqtt_client", server = "bipes.net.br", port = 1883, user = "bipes", password = "m8YLUr5uW3T");
>>> easymqtt_client.connect()
0
>>> easymqtt_client.publish('iysj9n/data/', 'sometestdata')
>>> 
```

I tried a lot but  i could not get the mqtt-session icon in the status bar to stop blinking. Also no data was displayed in the v3 dashboard.