BIPES/BIPES #139: LED Pin is not named as in previous version

wohltat, 2022-06-20:
[image]

Version 3.0.10

in Version 2:
[image]

gastmaier, 2022-06-21:
Humm, the WemosD1 mini seems to have this pinout
[image]
and the ESP8266
[image]
which matches what the third version is rendering for each device.

The file that sets the pinout is [static/page/device/devices.js](https://github.com/BIPES/BIPES/blob/third/static/page/device/devices.js)

wohltat, 2022-06-21:
The pin mapping is fine. It's not about differences between ESP
This issue is just about that is harder to find the onboard LED pin for a beginner. 
Since pins are labeled with SCL, SCA, etc. it would be nice to have this for LED as well.

I can try to add it, but i have some issues with my git setup i need to solve.

closed 2022-06-22 by wohltat