BIPES/BIPES #149: broken indentation when viewing generated code in the console

wohltat, 2022-09-19:
on `https://bipes.net.br/3/ide?page=prompt` i see this after starting a simple blink sketch 

```
Connected with WebSerial!

>>> import os; os.uname()
(sysname='esp8266', nodename='esp8266', release='2.2.0-dev(9422289)', version='v1.18 on 2022-01-17', machine='ESP module with ESP8266')
>>> import os; os.uname()
(sysname='esp8266', nodename='esp8266', release='2.2.0-dev(9422289)', version='v1.18 on 2022-01-17', machine='ESP module with ESP8266')
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import time
               from machine import Pin

                                      def gpio_set(pin,value):
                                                                if value >= 1:
                                                                                  Pin(pin, Pin.OUT).on()
                          else:
                                   Pin(pin, Pin.OUT).off()

                                                          for count in range(10):
     time.sleep(1)
                      gpio_set((2), True)
                                             time.sleep(1)
                                                              gpio_set((2), False)
```

wohltat, 2022-09-19:
Tested with micropython 1.18 and 1.17 (installed via BIPES web installer)

freemansoft, 2023-01-16:
Looks a lot like a line ending issue CRLF vs LF.