BIPES/BIPES #156: Missing blocks to execute arbitrary python code

wohltat, 2022-09-19:
In BIPES version 2 there was the possibility to run arbitrary python code.
[image]

In version 3 it is missing
[image]

This is a crucial element if you create a block language that will always miss some blocks.

Furthermore this is the only way to enter mathematical calculations without going crazy. 
Lets be honest, blocks can make programming easier, but entering calculations definitely gets more complicated.

-----------------
[BIPES version 3](https://bipes.net.br/3/ide-en)

wohltat, 2022-09-20:
For me this is the most important issue because there is no other way fix problems on the block side (automated without console). I think a 'run arbitrary code block' is essential.

Also it is an easy way to add or test new functionality.

freemansoft, 2023-01-16:
What branch is V3 on?

bikeNomad, 2023-04-07:
V3 appears to be on the `third` branch.