BIPES/BIPES #169: Servo position error invoking API - named parameter doesn't exist

freemansoft, 2023-01-15:
MicroPython v1.19.1-782-g699477d12 on 2022-12-20; Raspberry Pi Pico W with RP2040

```
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import PWM
=== from machine import Pin
=== 
=== 
=== pservo = Pin((0))
=== servo = PWM(pservo,freq=50)
=== servo.duty(90)
=== 
=== 
=== 
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
TypeError: function doesn't take keyword arguments
```

jetannenbaum, 2023-01-15:
Hi Joe,

   There are a few "quirks" in the RP204 implementation of MicroPython.
Try this version of BIPES is optimized for the Pico, Pico W, and MakerPi
RP2040: https://bipes.net.br/pico/ui/

   Thanks,

     Jet

freemansoft, 2023-01-15:
That version worked for the RP2040 with respect to servos.

freemansoft, 2023-01-16:
This is fixed in the offline branch

closed 2023-01-16 by freemansoft