BIPES/BIPES #174: servo # offsets ignored when initializing and using the servos

freemansoft, 2023-01-16:
branch: `offline`

Scenario
1. Initialize Servo 4
3. Move Servo 0

Expected Output
* No action or error

Actual Output
* Servo motion on servo 0 actually operate against servo 4

```
=== from machine import PWM
=== from machine import Pin
=== import time
=== 
=== def setServoAngle(servo, angle):
===  pulse_min_usec=500
===  pulse_max_usec=2500
===  timer_resolution=65535
===  pulse_range_usec=pulse_max_usec-pulse_min_usec
===  angle = max(min(angle, 90),-90) + 90
===  pulse_target = (angle * pulse_range_usec // 180) + pulse_min_usec
===  tics = pulse_target * timer_resolution // 20000
===  servo.duty_u16(tics)
=== 
=== 
=== pservo4 = Pin(4)
=== servo4 = PWM(pservo4)
=== servo4.freq(50)
=== setServoAngle(servo0,0)
=== time.sleep_ms(2000)
=== setServoAngle(servo0,90)
=== time.sleep_ms(2000)
=== setServoAngle(servo0,(-90))
=== 
```

PWM may have the same issue if the Servo code was copied from the PWM code

freemansoft, 2023-01-16:
Weird.  I just ran this against https://bipes.net.br/pico/ui/ and it correctly detected this issue

jetannenbaum, 2023-01-16:
Hi Joe,

   I couldn't reproduce.  Could be that your offline flow had the servo
defined offscreen.  Or something else wasn't clean when you were testing..
Generally, if something "odd"  happens I close the browser, reopen, and
restart BIPES.  Clearing the cache is also advised.

   Thanks,

     Jet

freemansoft, 2023-01-17:
Code changes have been made to address this

closed 2023-01-17 by freemansoft