BIPES/BIPES #176: How are libraries managed and loaded?  Why pylibBlobs and pylibs?

freemansoft, 2023-01-19:
Why does BIPES seem to have two ways to handle libraries?  Why does `pylibBlobs` exist when there is a `pylibs` method of adding them.  

There are raw `.py` file libraries in the `pylibs` directory and Javascript string wrapped libraries in `pylibBlobs`.  
* `pylibs.rst` describes how to add a `.py` library to the system using the `pylibs` directory. That is pretty clean.
* The most recent addition to the library was an LCD library that was put in `pylibBlobs`.  
* The recent addition of the lcd library from `pylibBlobs` didn't add it to code.js where it looks like the `pylibBlobs` are loaded.
* Some of the libraries exist in both places `pylibs` and in `pylibBlobs` EX: `songs`

I'm asking because libraries in `pylib` are cleaner without string wrapping and easier to test by copying and pasting in a repl.

jetannenbaum, 2023-01-19:
Good Morning Joe,

   The pylibBlob libraries exist for the offline version of BIPES.  You
can't load the files from the pylibs folder when running BIPES locally,
unless you change some security settings on the browser (which I do not
advise and am unwilling to do).  I maintain them as I test and try
different sensors, displays, etc.  In some cases, I am waiting for parts
and try to get ahead by creating the pylibBlob file before the part shows
up.  The lcd library files are an example of that.  I added the libraries
without finishing the rest of the changes required to make them available.
I do _not_ add those libraries to the main branch (or the pylibs folder)
simply because the main branch code mostly works for the ESP boards.  I
only work with RP2040 boards.  So I have no way to test the library files
with other boards.

   Thanks,

     Jet

freemansoft, 2023-01-19:
This is great info.  Thanks.  So pylib and pylibBlob exist because of the different deployment models. That is super interesting.  I may try and do a PR doc update to capture that if you’re ok with it.

freemansoft, 2023-01-19:
Should the `pylibBlob` files be code generated from the `pylib` folder? It could be added to the Makefile.  Then the files could be maintained in one place.

jetannenbaum, 2023-01-19:
Sorry,

   That doesn't work.  Some of the libraries are different, even though
they have the same names.  The pylibBlob files are specifically tested for
the RP2040 boards.   MicroPython has some differences between the RP2040
implementation and other non-RP2040 boards.  For example, the SSD1306
library in the pylib folder is a different implementation than the SSD1306
library in the pylibBlob folder.  I used the SSD1306 library that is
available through Thonny's package manager as the basis for the file in the
pylibBlob folder

   Thanks for the offer,

     Jet

closed 2023-01-19 by freemansoft