BIPES/BIPES #179: Is this project dead?

wohltat, 2023-01-26:
Is there anyone still working on this project?

jetannenbaum, 2023-01-27:
Hi Christian,

   I maintain the offline branch of the project.  It is centered on the
RP2040 Microcontroller, since those are the only microcontrollers I own.
Not sure if I can help you, but if you have a question, I can try.

   Thanks,

     Jim

wohltat, 2023-01-27:
Hi,
i've opened some issues around 4 moths ago, but so far no one has answered them. Maybe you have some ideas on them.

Is the offline branch basically a version 2 ide without a server? Is there a way to make MQTT work with it?

jetannenbaum, 2023-01-28:
Hi Christian,I believe the issues you opened related to a beta version that others had created.   Sorry, not much help from me on those.  I haven’t looked at MQTT in the BIPES code, but happy to take a look this weekend.   I’ll report back with any updates tomorrow evening JetSent from Jet’s iPhone

jetannenbaum, 2023-01-28:
Hi Christian,

I found the issue with EasyMQTT.  There is a library that needs to be loaded.  I haven’t looked at the MQTT code yet.  Still trying to figure if I can run EasyMQTT inside of offline BIPES.  I did get it to run from offline BIPES to a second browser window running EasyMQTT, but that isn’t ideal.  Stay tuned for updates tomorrow 

Jet

jetannenbaum, 2023-01-29:
Hi Christian,

    The issue is that the mqtt libraries are not loaded.  I haven't figured our how to load them in the proper directory structure.  I can load them using Thonny, but not BIPES.  Also, I can run EasyMQTT in the offline version of the code by starting EasyMQTT in a new window.  Starting it with offline BIPES causes a cross scripting error.  This is going to take more research for me to resolve.  The easiest solution at the moment for you to download the libs to you device until I can figure out a clean way to get them there.  The libs are found here: https://github.com/micropython/micropython-lib/tree/master/micropython  You will need two libs: umqtt.simple and umqtt.robust.  Make sure to put them in the umqtt folder on your device.  I'll keep working on a solution that gives a button to download the files, but haven't gotten it to work with the correct directories yet.

Thanks,

 Jet

jetannenbaum, 2023-02-04:
Hi Christian,  

   I finally got back to this issue and fixed the code for the Pico W.  Take a look at https://bipes.net.br/pico/ui/.  Make sure to load both the simple and robust libraries.  I only tested with EasyMQTT.  I have not looked at the MQTT code. 

Thanks,

Jet

jetannenbaum, 2023-02-04:
Hi again,

   I tested the MQTT code and it is confirmed to be working as well.  If you have any issues, please clear your browser.  If that doesn't resolve the issue, let me know. 

   Thanks,

    Jet

sanyaade-teachings, 2023-02-10:
> Hi Christian, I maintain the offline branch of the project. It is centered on the RP2040 Microcontroller, since those are the only microcontrollers I own. Not sure if I can help you, but if you have a question, I can try. Thanks, Jim
> […](#)

@jetannenbaum 

Hi Jetannenbaum, 

Good day to you and thanks for maintaing the RP2040 Microcontroller part of this project.

I am interested in the RP2040 Microcontroller, we are using Cytron Maker RP2040 for Robotics in our community.

Please is it possible to outline on how to get it build locally for RP2040 Microcontroller? Many thanks!

I tried to build it many time but the RP2040 Microcontroller is not added to BIPES device selection and not showing/being loaded.

What I am doing wrong?

Please assist us. Give steps to build BIPES, build the Offline branch that have RP2040 enabled/loaded.

Many thanks!

Hear from you soon!

God blesses!!!

Regards

jetannenbaum, 2023-02-10:
Hello Sanyaade,

You do not need to build the offline version.  You can access the version online that supports the Maker Pi RP2040 Board here: https://bipes.net.br/pico/ui/

If you must run locally, you need to do the following from a command prompt:
1. Clone the BIPES Repository to your local drive: git clone https://github.com/BIPES/BIPES
2. Change directory to BIPES: cd BIPES
3. Next, point to the offline branch: git checkout offline
4. Run the makefile:
   a. In Linux, it should be bash Makefile
   b. I run on a windows laptop and use cygwin to run my makefile, so the command would be bash cygwinOffline.mak
5. Change directory to ui: cd ui
6. There is a small bug in the cygwinOffline.mak file that makes the index_offline.html file.  There is an unprintable character after the last letter in the file name.  Simply delete it.  Once you do, the file should be loadable from a browser.
7. That's it.

sanyaade-teachings, 2023-02-10:
Hi jetannenbaum,

Many thanks form your quick reply.

Yes! I have to run it locally due to internet connection issues within the community premise.

I am adding it as module to internet-in-a-box (http://iiab.net/home/) which runs in Ubuntu Linux box

The kids laptops(Windows based) and Chromebooks will access BIPES through our local network.

Many thanks! Following your outline steps, I managed to build it and run it on. 

It works as expected. I need to test the connection to a Windows laptop and Chromebooks. 

Also see how to auto-connect  to a Maker Pi RP2040 on Linux box as well.

I will be looking into how to adapt CoderDojo Twin City learning resource for RP2040. Many thanks to you guys for putting that resource on the internet. :+1: 

Once again, many thanks for your quickly reply and support. Much appreciated!

God blesses!!!

Regards

samuk, 2023-09-30:
I was wondering the same. Was it a grant funded academic projects?

bonalex01dev, 2023-11-19:
Hi I would like the project to be updated to the latest micropython version. How can I help for this? Anyway thanks for this wonderful tool!

jetannenbaum, 2023-11-19:
Which version of BIPES are you using?

bonalex01dev, 2023-12-12:
Hi, I'm using the 0.2.3 offline version 
in ESP32 mode.
The device is a ESP32 S3 with spiram

Maybe there's not so much difference with the online version ? Though I noted the Python "try except" may come in handy and is not present in the 0.2.3.

Anyway thanks for this wonderful piece of code!

jetannenbaum, 2023-12-18:
The try except block is included in the Python menu.

[image]

bonalex01dev, 2023-12-19:
Hi,
thanks but I can see this block in the online version but not in the v0.2.3 from the github release page for offline use

If I download the main github files, I can't use them offline. Here's what I tried
- download / unzip github repo
- run index.html (fail)
- run UI/index.html (fail)
- copy and run UI/index_offline.html from release version of BIPES. (better, I obtain a BIPES UI but without any block menu, interface unusable)

May I suggest you could publish a newer version in the releases... with the updated offline html files.

Anyway your tool is great!

jetannenbaum, 2023-12-20:
Do you clone the github repo?  If so, you need to switch to the offline
branch.  Do this:

git clone https://github.com/BIPES/BIPES  (That will get the main branch)
git checkout offline
you will need to build the offline version.  Depending on your O/S, you can
either run the makefile directly or use cygwin to build the
offline_index.html file.  I use a custom version of the makefile under
cygwin, since I'm only interested in the RP2040 boards and I run on a
windows machine.

Jim

bonalex01dev, 2023-12-23:
thanks it works great!

rafaelaroca, 2024-11-10:
Closing this issue as we can discuss each topic with a name different than "is this project dead" :-)

closed 2024-11-10 by rafaelaroca

wohltat, 2024-11-14:
I'm happy to see that there seems to happen a lot in the project lately. This makes me more motivated to participate again.
I hope that this project will become more mature and stable because i really like it and would want to use it more with our students at school.