BIPES/BIPES #181: Self-Hosted(Localhost) HTTPS issues

sanyaade-teachings, 2023-02-12:
Hi All,

Good day top you all wherever you are!

We are having issues with a local hosted BIPES for community.

The error is as shown below:
[image]

The web serial link will move to a new online page and not a local web serial.
Question: 

1. How can stack a local HTTPS edition of BIPES?
2. BIPES folders/directory layout: where can we find information about this?

We are using the Maker Pi RP2040 kit currently with the kids
Internet has been issues hence things have to run on a local server (we have access points) for kids to access BIPES

Many thanks for all your  support and efforts. Much appreciated! 

God blesses!!!

Regards

jetannenbaum, 2023-02-12:
Hi Sanyaade,You cannot connect to the Maker Pi board that way.  Use the serial connection.  You will need a usb cable to connect your computer to the boardI don’t understand your question about the folders and layout.  Jim Sent from Jet’s iPhone

closed 2023-02-23 by jetannenbaum

sanyaade-teachings, 2023-02-26:
@jetannenbaum :

Sorry getting back here late due to internet connection issues at our ends here.

Also, many thanks for your reply. 

We are connecting the board through USB cables but the kids browsers (e.g. Chrome/Firefox) does not allow connection through the web-socket. I am trying to see if we get past the issues of https as required by most browsers nowadays.

Anyway, we got this issue solved by using mkcert ( https://web.dev/how-to-use-local-https/ ) to create a **https://localhost**. This get the internet browsers working for the learners so far. 

I need to look a bit more deep into how BIPES is structure. Also it looks like you are the only active developer currently.

Once again many thanks for your time and support. Much appreciated!

God blessings!!!

Regards