BIPES/BIPES #189: Generating different code using "https://bipes.net.br/ide/" and offline.

LCKidsLabs-01, 2023-02-20:
I do afterschool programs so I don't always have a solid internet connection. That is why I stared using BIPES.  

On a Raspberry PI Pico using "Start HCSR04 Ultrasound sensor". The online version works but the offline version doesn't work. It was one of my students that pointed this out to me. We are using our a lot of "Run Python Code" blocks to make our ultrasound sensors work offline.

**Online at "https://bipes.net.br/ide/" we get:**

```python
#Based on: https://github.com/lemariva/uPySensors
class HCSR04:
    def __init__(self, trigger_pin, echo_pin, echo_timeout_us=500*2*30):
        self.echo_timeout_us = echo_timeout_us
        self.trigger = Pin(trigger_pin, mode=Pin.OUT)
        self.trigger.value(0)
        self.echo = Pin(echo_pin, mode=Pin.IN)

    def _send_pulse_and_wait(self):
        self.trigger.value(0)
        time.sleep_us(5)
        self.trigger.value(1)
        time.sleep_us(10)
        self.trigger.value(0)
        try:
            pulse_time = time_pulse_us(self.echo, 1, self.echo_timeout_us)
            return pulse_time
        except OSError as ex:
            if ex.args[0] == 110: # 110 = ETIMEDOUT
                raise OSError('Out of range')
            raise ex

    def distance_mm(self):
        pulse_time = self._send_pulse_and_wait()
        mm = pulse_time * 100 // 582
        return mm

    def distance_cm(self):
        pulse_time = self._send_pulse_and_wait()
        cms = (pulse_time / 2) / 29.1
        return cms
```

**Offline we get:**
```python
from hcsr04 import HCSR04

ultraSoundSensor = HCSR04(trigger_pin=(25), echo_pin=(25), echo_timeout_us=10000)
```

jetannenbaum, 2023-02-20:
Hi,

   The offline version uses a library, rather than the code that the online version uses (it is probably the same underlying code).  However, you still need to have different pins for the echo and trigger.  Thanks for trying the offline version.

LCKidsLabs-01, 2023-02-21:
If you don't mind me asking, what is the reason for difference code base.  Yes, I was slightly careless in pin assignment but I was just trying to give an examples of the differences not actually show working code.

jetannenbaum, 2023-02-21:
I built a version of the code for the RP2040 boards that ran offline.   The online base code and offline code base is different for two reasons:
1. I only have RP2040 based boards, so I can only test that code.   I don’t want to contaminate the main code base.
2.  You can’t download libraries from the internet when you run the code locally without opening up your system to potential hackers.   You have to disable the browser security settings.

Once I contacted the original author, he allowed me to use the site to create an online version of my code base.   My online code is a revision ahead, since I haven’t had time to sync up 

Sent from Jet's iPad

closed 2023-02-23 by jetannenbaum