BIPES/BIPES #190: On the RP2040 based boards BIPES is using wrong syntax for timers

LCKidsLabs-01, 2023-02-20:
There is a function started but is not complete called pico_timer. I would remove the this block because it is not used and update the code as below if that is OK with you. 

It seems to me that we should just change syntax
https://github.com/BIPES/BIPES/blob/master/ui/core/generator_stubs.js#L4459
**form:**
```javascript
Blockly.Python.definitions_[`import_timer_start${timerNumber}`] = `tim${timerNumber} = Timer(${timerNumber})`;
```
**to:**
```javascript
if (UI ['workspace'].selector.value == "Raspberry Pi Pico" || UI ['workspace'].selector.value == "Raspberry Pi Pico W") {
    Blockly.Python.definitions_[`import_timer_start${timerNumber}`] = `tim${timerNumber} = Timer()`;
} else {    
    Blockly.Python.definitions_[`import_timer_start${timerNumber}`] = `tim${timerNumber} = Timer(${timerNumber})`;
}
```

**Then remove** 

https://github.com/BIPES/BIPES/blob/master/ui/core/generator_stubs.js#L4468
```javascript
Blockly.Python['pico_timer'] = function(block) {

  var interval = block.getFieldValue('interval');
  var timerNumber = block.getFieldValue('timerNumber');
  var statements_name = Blockly.Python.statementToCode(block, 'statements');
  
  Blockly.Python.definitions_['import_timer'] = 'from machine import Timer';
  Blockly.Python.definitions_['import_timer_start'] = 'tim=Timer()'; //-1)';

  Blockly.Python.definitions_['import_timer_callback'] = '\n#Timer Function Callback\ndef timerFunc(t):\n' + statements_name + '\n\n'; 

  var code = 'tim.init(period=' + interval + ', mode=Timer.PERIODIC, callback=timerFunc)\n';
             
  return code;
};
```

and remove 

https://github.com/BIPES/BIPES/blob/master/ui/core/block_definitions.js#L455
```javascript
Blockly.Blocks['pico_timer'] = {
  init: function() {
    this.setColour(135);
    this.appendDummyInput()
        .appendField("RPI Pico Timer ")
        .appendField(" Interval (ms): ")
        .appendField(new Blockly.FieldTextInput("1000"), "interval");
    this.appendStatementInput("statements")
        .setCheck("image");
    this.setPreviousStatement(true);
    this.setNextStatement(true);
    this.setTooltip('');
  }
};
```

jetannenbaum, 2023-02-21:
The RP2040 based boards can have many timers; however, the deinit method will stop all timers.  I'm not sure if it is wise to create multiple timers in this manner for the RP2040 based boards.  I'm willing to consider the change; but will need some convincing.

LCKidsLabs-01, 2023-02-21:
I am not sure I can convince you since I don't understand your comment, Below is an example of the way I use deinit on a RP2040 running "MicroPython v1.19.1-878-g4598b89ce on 2023-02-17; Raspberry Pi Pico W with RP2040". When I run the code below the board LED stops blinking after 15 sec but the redLED keeps blinking.

```python
from machine import Pin,Timer
import time

led = machine.Pin("LED", Pin.OUT)
redLED = Pin(14, Pin.OUT)
tim0 = Timer()
tim1 = Timer()

#Timer Function Callback
def timerFunc0(t):
    led.toggle()

#Timer Function Callback
def timerFunc1(t):
    redLED.toggle()

tim0.init(period=1000, mode=Timer.PERIODIC, callback=timerFunc0)
tim1.init(period=1000, mode=Timer.PERIODIC, callback=timerFunc1)
for count in range(15):
    time.sleep(1)

tim0.deinit()
```

jetannenbaum, 2023-02-21:
You convinced me :-).   I’ll have to expand your suggested solution to include the Maker Pi RP2040 board.   I’ll add it in in a few days.  I’ve been away from the code for a few days and need some time to catch up

Sent from Jet's iPad

LCKidsLabs-01, 2023-02-21:
I will make up a PR for you, and include the Maker Pi RP2040 board. We should also start adding some Adafruit boards in the near future.

LCKidsLabs-01, 2023-02-21:
So the PR should be to Offline and not Main is that correct?

jetannenbaum, 2023-02-21:
Please wait until I get the offline code in sync with the online Pico code.  I have a few updates to add.   And yes, if you do a PR, please do so against the offline code if you would like me to incorporate it.

Thanks

Sent from Jet's iPad

jetannenbaum, 2023-02-21:
Hmmm, I don’t have Adafruit boards.  Which boards do you recommend?

Sent from Jet's iPad

LCKidsLabs-01, 2023-02-21:
I have a few but I would say the Feather is a good starts. I like this board because it is small and there is a large ecosystem for the feather platform. 
https://www.adafruit.com/product/4884

jetannenbaum, 2023-02-21:
Interesting board.  Not widely available near me.  There is only one in stock at the nearest Micro Center.   I’ll try to pick it up.Sent from Jet’s iPhone

https://www.adafruit.com/product/4884

LCKidsLabs-01, 2023-02-21:
Go out to my website https://www.LCKidsLabs.com and find my contact info. Send me your contact info I will get one sent out to you.

jetannenbaum, 2023-02-21:
Thank you I sent my contact information Sent from Jet’s iPhone

LCKidsLabs-01, 2023-02-22:
Adafruit is just about to rerelease there rp2040 Feather. Here is a video from yesterday:
https://www.youtube.com/watch?v=oNi_D_TUs88
It show the good work they are doing.  I think after seeing this we should wait a month to see what comes out and just start supporting the newest stuff going forward. I have a few of there current RP2040 boards but most of them are in my little ones project.

jetannenbaum, 2023-02-23:
I resolved the issue with the timer code.  No reason to do a PR.  The offline code is now in sync with my online pico code.  I updated the RP2040 toolbox files to call the timer functions rather than the pico_timer functions.

closed 2023-02-23 by jetannenbaum

LCKidsLabs-01, 2023-02-24:
I gave it a few days to comment back, because frankly I was a little upset. People create pull requests on an open source project like this for a multitude of reasons.

My reasons are:
1) Betterment of the world in general
2) Fun
3) Credibility in our field
4) General sense of good feeling when you look at a repository and see you did work on it.

You took my code from issue https://github.com/BIPES/BIPES/issues/190#issue-1591015535 and pasted it after asking me to hold off on a PR!

> Please wait until I get the offline code in sync with the online Pico code. I have a few updates to add. And yes, if you do a PR, please do so against the offline code if you would like me to incorporate it.

It took me 3 days to fully understand the flow of the project. If this is the way this project is normally run you are not going to get much repetitive contributions.

I will be posting a similar response in PR https://github.com/BIPES/BIPES/pull/187#issue-1590406563 as you did the same thing.

jetannenbaum, 2023-02-25:
> You took my code from issue #190 (comment)
> I will be posting a similar response in PR #187 (comment)

That's a fair criticism and I apologize.  I looked at your code and was
enthusiastic to get it working.  In the future, I'll hold off for a bit
until the PR is complete and we can discuss any issues.

  Thanks for the feedback