BIPES/BIPES #209: JSON and dictionaries

QuirkyCort, 2023-09-05:
Suggest to have a set of blocks for dictionaries, as well as json dumps/loads. You can see IoTy for an example.

This will allow the user to extract data from public API (eg. https://github.com/public-apis/public-apis). An example project would be to retrieve and display the weather.

jetannenbaum, 2023-09-12:
This gets a bit complicated.  The Python block may do what you want, since it exists to allow a user to add code if there aren't blocks to support the desired function.  Please take a look at that block and let me know if that solves the problem.  Thanks for supporting BIPES.

QuirkyCort, 2023-09-12:
Certainly, the Python block will do the job; it should be able to replace any blocks.

But dictionaries is a basic type, and JSON is commonly used in public APIs, so I think it would make sense to support it in blocks.