BIPES/BIPES #211: Blocks for GY33 and TCS3472

QuirkyCort, 2023-09-05:
The TCS3472 is a color sensor (...I think it's the one you showed me when we met on zoom). The GY33 uses the TCS3472, but contains a microcontroller that provides it's own UART and I2C output. Both are good for robocup line following / color detection, but the GY33 has the advantage of having a shroud that minimizes interference from ambient light.

I couldn't find a GY33 lib in micropython, so I've written a pair (UART and I2C). https://github.com/QuirkyCort/micropython-gy33

For the TCS3472, there is an existing library by Rob Hamerling (https://gitlab.com/robhamerling/micropython-tcs34725), but I've also written my own (https://github.com/QuirkyCort/IoTy/blob/main/public/extensions/tcs3472.py).

You might want to include them into BIPES.

jetannenbaum, 2023-09-12:
Hi,
   I just bought the sensor and will be testing out your code over the next weekend.  Assuming I get time, I'll add it to the library.  Thank you for your efforts supporting BIPES.

jetannenbaum, 2023-12-19:
Hi,

   I finally got around to ordering the part and testing it (first,
directly with the RP2040 and MicroPython).  Using your serial driver, I can
turn the LED on & off; however, I can't get any meaningful data out of the
device.  Any suggestions?

The code:
from machine import UART, Pin
from gy33_uart import GY33_UART

uart0 = UART(0, baudrate=9600, tx=Pin(0), rx=Pin(1))
gy33 = GY33_UART(uart0)
#gy33.set_led(10, True)

gy33.update(wait=False, timeout=1000)
print(gy33.get_lcc())
print(gy33.get_calibrated())

Output:
[0, 0, 0]
[-15, -22, -22, -21]

Jim

jetannenbaum, 2023-12-20:
Ok,

  FIgured it out.  It helps to read all the documentation, although I'm not
sure what the additional pin (p0) is connected to..  I have the UART code
functioning.  Not sure if it makes sense to try the I2C code.

Thoughts?

  Jim

QuirkyCort, 2023-12-21:
I don't have a pin p0 on mine, are you referring to s0? If so, that pin needs to be grounded if you want the built-in MCU to operate in I2C mode. If left floating, it'll operate in UART mode.

There are some benefits to the I2C mode...

- Save a couple of pins if you are already using other I2C devices
- Avoid the need to run update(); in UART mode, you may end up with stale data if you don't run update() often enough

When working with students, I think the I2C mode is easier to use. One major drawback of I2C mode is that you cannot set the integration time. And on my device, the default integration time is too slow for applications like robotics line following. You can set the integration time in UART mode, and the setting will persist even if you power it off and switch to I2C mode, so a workaround would be to pre-configure the integration time to a more reasonable default before use.

jetannenbaum, 2023-12-22:
This is the UART code from your example with the po I referred to:

import machineimport gy33_uartimport time
p0 = Pin(0, Pin.IN, Pin.PULL_UP)
uart = machine.UART(1, baudrate=9600, tx=4, rx=5)gy33 =
gy33_uart.GY33_UART(uart)
gy33.set_led(10) # Full powergy33.set_output(True, False, True) #
Continuously output raw and processed RGB values
while True:
    gy33.update()
    if p0.value() == 0:
        print(gy33.get_raw())
        print(gy33.get_processed())
        time.sleep(1)

I couldn't get the I2C code to work at all.  Eventually, I used the
UART code to set the I2C addr.

  I can scan the address back using the i2c.scan(), but the device
gives me an EIO error.

Thanks,

Jim

QuirkyCort, 2023-12-23:
On the ESP32, the p0 is often connected to the "Boot" button and pulled high. In the UART example code, update() is running continuously without delay, but the values are only printed when the boot button is pressed.

In I2C mode you got the address on scan but couldn't read data? Try lowering the speed or adding an external pull up. I didn't need one on mine, but I'm using an ESP32 which may have a stronger internal pull-up than the RP2040.

jetannenbaum, 2023-12-23:
Well, you gave me a hint, since you noted that you are using the ESP32
chip, I figured that I needed to use 5 VDC to power the chip, not 3.3VDC
from the Pico.  No idea why the chip works fine with the UART at 3.3 VDC,
but so be it.  I tried 5VDC with the I2C code.  I can set the led (BTW, the
code example on github uses set_led_power instead of set_led) and I can
change the intensity of the led (off -> full power).  That said, I get
garbage when I try to read the data (again, the sample code uses read_cal
instead of read_calibrated).  If I put the chip into tcs34725 mode,
everything works with 5VDC.  Once I reset the chip back to "gy33 I2C mode"
I got reasonable readings.  Weird, but it seems to work.  I'll play with it
some more, add the I2C and UART code to the ESP32 base; however, until I
get more comfortable with the I2C code and the Pico, I'll only add the uart
code for the RP2040 boards.

Thanks for your help.  More updates to come as I get some time over the
holiday to update the code.

Happy Holidays!

Jim

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

QuirkyCort, 2023-12-24:
Pretty sure I was using 3.3V with the ESP32.

I still suspect it is a pull-up issue. If the GY33 uses an internal pull-up for I2C, then the 5V would have the effect of pulling the signal up faster than with 3.3V.

Can't begin to even guess why switching to TCS34725 mode would make a difference. Could be merely a coincidence.

jetannenbaum, 2023-12-25:
I played some more.  I have the code (slightly modified) running on my
Pico.  I tried pullup resistors, but no joy.  The device responds
appropriately (today) via I2C at 3.3 VDC.  The UART gives me data back, but
it doesn't scale.  I went back to the original class code on the site to
make sure I didn't see any differences in the output between my modified
UART class and your original code.  I'll put the code up tomorrow (12/26)
for both the RP2040 boards as well as the ESP32 boards.   You'll be able to
play with the code on the https://bipes.net.br/pico/ui/ site  Or you can
download the code locally and poke around in the offline branch.

Thanks for your help.  Happy holidays

  Jim

jetannenbaum, 2023-12-26:
The code is updated on the site.  Use the ESP32 Basic device.  There is a
bug in the ESP32.xml file that I need to fix before the module will show
up.  I don't have that device, so I haven't spent any time fixing the bug
in the GPS sensor section.  Something for another day.

 Jim

closed 2024-06-20 by jetannenbaum