BIPES/BIPES #217: Raspberry Pi PICO and ST7789 display

bobricius, 2023-10-10:
Hi is possible to add this display to Pico boards?

jetannenbaum, 2024-01-02:
Hi,

   It took too long, but I finally added this display to the pico version
of BIPES.  The code requires the Russ Hughes UF2 file, since it contains
the fonts needed for the display.  The link to his UF2 files are noted at
the top of the initialization for RP2040 boards.  The version of BIPES with
the display added can be found here:
https://bipes.net.br/pico/ui/

  I couldn't get the cheap TFT boards to work (the ones without chip
select).  I tested the code with the WaveShare displays.  Please let me
know if you have any questions or issues.

bobricius, 2024-01-02:
Amazing, Thanks very much
Working ...
[test preview
](https://twitter.com/bobricius/status/1742286629845250518)

jetannenbaum, 2024-01-02:
Thanks for the link that shows it working for you.  Jim

closed 2024-06-20 by jetannenbaum