BIPES/BIPES #224: MicroPython Machine.Pin uses incorrect Import sytax

sk8board, 2024-03-27:
The blocks `micropython > machine.Pin > Pin.on and Pin.off` use the following library call syntax `import machine.Pin`. This syntax should be `from machine import Pin`.

[image]

The bug appears to be in the following lines of code

https://github.com/BIPES/BIPES/blob/master/ui/core/generator_stubs.js#L1940

and

https://github.com/BIPES/BIPES/blob/master/ui/core/generator_stubs.js#L1946

>```js
> Blockly.Python.definitions_['import_machine.Pin'] = 'import machine.Pin';
> ```

[image]

This bug results prevents the code from running.

[image]

sk8board, 2024-03-27:
@rafaelaroca @jetannenbaum I noticed development with this project stopped a few years ago. This looks like a very good project that needs to be finished. What caused development to stop?

jetannenbaum, 2024-03-27:
Hi,

   I support the offline version of the code and the RP2040 processor.  I made an update yesterday to the TFT display to support circle and circle fill.  As far as changing the MicroPython section of the code, I took a look and it is more involved than just the machine.Pin section.  It appears that most of the machine import code is suspect.  Perhaps that style of import was supported in a previous MicroPython build.  If I may, why are you using that code instead of the Machine In/Out Pins blocks?

   Thanks,

    Jim

jetannenbaum, 2024-03-27:
[image]

sk8board, 2024-03-27:
Thank you for your help!

I have been looking for a tool to help my son use his understanding of Scratch to develop an understanding of MicroPython with a Pi Pico RP2040. I recently came across BIPES and wanted to test various blocks to see how well BIPES functions. That is why I was using the MicroPython section with machine.Pin.

I agree there are many blocks of BIPES that have problems with an old version of MicroPython. That is why I thought development stopped. How much work do you expect it would take to get the many blocks updated to a recent version of MicroPython?

Also, I am using the Web version and keep running into a problem with the Web Serial API where the UI will freeze after connecting. Is it better to use an offline version?

[image]

jetannenbaum, 2024-03-27:
I tend to use the offline version for testing, before I push the updates.  Just to be clear, I'm maintaining the version of BIPES that can be found here: https://bipes.net.br/pico/ui/

Interesting that you are choosing to use the MicroPython blocks rather than the generic blocks.  The generic blocks will generate the appropriate MicroPython code.  I'm not sure how long it would take to fix all of the MicroPython blocks, since I never use them.  Do you have a specific set of blocks that you want to use?

sk8board, 2024-03-27:
I tried the online version that you maintain and I cannot get it to run the default blink sketch. If I select Run Block Based Program button on the Console, I get the following error. I have tired this with two different computers and two different Pico's with the latest version of MicroPython binary file.

[image]

[image]

jetannenbaum, 2024-03-27:
shoot me a dm jetannenbaum@gmail.com and I'll send you a zoom link.  We can look at together.

sk8board, 2024-04-14:
Closing this issue because the MicroPython blocks are not developed and there is not a plan for developing the MicroPython blocks.

closed 2024-04-14 by sk8board