BIPES/BIPES #227: LCD 16x02 Character Display Code Block Assumes I2C 0x27 Address

sk8board, 2024-04-08:
The LCD 16x02 character display code block assumes the I2C address is 0x27. If your display has a different address than 0x27, then the LCD block will not function for your display.

The LCD code block needs to allow the user to change the I2C address as needed. Therefore, the LCD block would need to add another input for I2C address.

With that said, there should be an additional code block that can scan for I2C addresses so the user can find the address of their LCD display.

[image]

[image]

jetannenbaum, 2024-04-08:
@sk8board: Do you have a specific board in mind that uses a different address?  Also, adding an I2C scan() method, while trivial, gets us right back into the discussion of the I2C init(...) method.  

From what I can see, it looks like one of the libraries (lcd_api) isn't loaded.  Did you load both the lcd_api & pico_i2c_lcd libraries?

sk8board, 2024-04-08:
@jetannenbaum Yes, I loaded both libraries. The problem is my display has an address of 0x3F but the block assumes 0x27.

jetannenbaum, 2024-04-08:
@sk8board:  Seems like the code should scan for the address, rather than assume that the address is 0x27.  That would be a trivial change inside the I2C Character LCD code (char_lcd_init block).  Want to take a swing at modifying the code?

sk8board, 2024-04-08:
@jetannenbaum I am attempting to fix this by adding another input field which would allow the user to choose a different address. 

Unfortunately, I discovered Blockly does not support hexadecimal numbers with their number fields. Therefore, I tried to use a text field, but a text field looks out of place as well as the text variable gets imported into MicroPython with '3F ' apostrophe marks.

[image]

[image]

I like your idea. I will give it a try.

sk8board, 2024-04-08:
@jetannenbaum I coded both options and I prefer the dropdown list.

I will submit a pull request and link it to this issue.

closed 2024-04-09 by sk8board