BIPES/BIPES #229: Console code does not paste accurately

sk8board, 2024-04-10:
@jetannenbaum I have experienced a problem of random errors in the console as shown below. 

This seem like code is being pasted with errors in the console. I have verified the code generator is showing correct code, but it seem like the automated copy and paste function of BIPES is malfunctioning at random.

I find this problem occurs more frequently with larger block programs. Sometime, with very large block programs, I rarely get the code to run without errors. Each time I run the code, the error occurs on a different line of code.

Do you know where to start with troubleshooting this problem?

[image]

[image]

sk8board, 2024-04-10:
Run button executes `Tool.runPython()`

[image]

`Tool.runPython()` executes  

```js
      mux.bufferPush (`\x05${code}\x04`);
      UI ['progress'].start(Channel.websocket.buffer_.length);
```

[image]

sk8board, 2024-04-10:
[image]

sk8board, 2024-04-10:
https://github.com/BIPES/BIPES/blob/fec9f9ef44c0f0a91d26a8b5ef782228e07a8bd4/docs/core/channel.rst

jetannenbaum, 2024-04-10:
Hi, this is not a problem that I have experience with, mostly because it isn't how I use BIPES.  My guess it is it is caused by a buffer overflow, but I would have to dig deep into the code to find it and make it happen so it could be fixed.  How are you attempting to debug?  Since I use Chrome, I would fire up the dev tools (ctrl+shift+i) and watch for errors in the console as the code in question executes.

sk8board, 2024-04-10:
I assumed it was buffer overflow related with "length" being incorrectly calculated.

```js
      mux.bufferPush (`\x05${code}\x04`);
      UI ['progress'].start(Channel.websocket.buffer_.length);
```
I do not know where to start with troubleshooting web browser buffer issues. I tried (ctrl+shift+i) as you recommended but the console does not show anything (literally nothing).

Here is a long program that is problematic.

[LCDdiscussion.bipes.zip](https://github.com/BIPES/BIPES/files/14934969/LCDdiscussion.bipes.zip)

jetannenbaum, 2024-04-10:
I loaded the program and ran it without issue.  I'm doing so locally, running the Chrome browser and on a Pico.  Are you saving the program onto your device and then trying to load it? I did have to fix the code to get it to run (freqen wasn't defined, change the processor to Pico, and change the pin assignments).

sk8board, 2024-04-10:
I think we have a different user experience because we are using BIPES differently. 

Here is the process that I thought was the design intent. Am I missing something?

[image]

Here is the result.

[image]

jetannenbaum, 2024-04-11:
I tried to follow the steps to build your code the way you did; however, I still cannot replicate the error that you get.

sk8board, 2024-04-11:
My son has been enjoying BIPES with an ESP32. He has our electronics bench covered with this setup for testing every block of BIPES, with an ESP32. Which is why he is finding occasional bugs.

Since I did not want to disturb his setup, I verified the problem with his ESP32 using two different computers as well as with Chrome and Edge on both computers. I can get this problem to occur with consistency.

Since you could not duplicate this problem, I disconnected his ESP32 setup and connected a Pico. I agree with you, I could not duplicate the problem with a Pico.

This appears to be a problem with BIPES and the ESP32.

jetannenbaum, 2024-04-11:
A few thoughts...   Since this seems to be hardware dependent, I would guess the issue is related to how the ESP32 handles filling the requested buffer.  You could make the buffer larger.  I don't see any reason why not, other than perhaps performance.

You could make the program simpler by using lists and functions.  That would help by reducing the "cut and paste" aspect of the code.

Debugging...  Once you open the dev tools in chrome, click on the Sources tab.  From there, you can load the code that you want to trace and set breakpoints, look at the call stack, and watch individual variables.

sk8board, 2024-04-11:
> Debugging... Once you open the dev tools in chrome, click on the Sources tab. From there, you can load the code that you want to trace and set breakpoints, look at the call stack, and watch individual variables.

Thank you. This was helpful.

I found if I placed a breakpoint at line 497 of `channel.js`, then the problem does not occur. The breakpoint is at `serialWrite(data)` which shows the array of data when the breakpoint is triggered. The buffer holds 475 characters at 8 bits each.

If I disable breakpoint diagnostics, the problem reoccurs. I am not sure how to diagnose this problem when diagnostic tools prevent the problem from occurring. 

[image]

jetannenbaum, 2024-04-11:
That makes me think it is a timing issue between the board and the interface.  Not sure what to do at this point, since you can't use the debugger to examine the problem.  Have you tried increasing the buffer size or adding a short delay between data reads?

sk8board, 2024-04-11:
Thank you for your help. I was able to find a solution that works.

Changing the buffer size did not work, but adding a 100ms delay did work.

ESP32 uses a CP2102N USB to UART serial converter. I read in the Serial Communication Guide for CP210x that a 60ms delay is "good practice".

I am going to have my son test it for a few days before submitting a pull request.

For future reference, buffer size is set for each board in `devinfo.json` using the variable `serial_packet_size`.

[image]

QuirkyCort, 2024-04-12:
FYI. pyboard.py uses a 256 chunk size with a 10ms delay between each chunk in raw mode. The micropython documentation suggests that you can send up to 256 bytes at a time, but that's referring to raw paste mode, so I'm not sure if that's applicable to raw mode.

Anway, I would assume that the pyboard.py people know what they are doing, so IoTy follows the same approach. Haven't had any issues with serial transfer for either pyboard.py or IoTy.

sk8board, 2024-04-12:
Thank you for your input.

I found the CP2102N USB to UART bridge has a buffer of 512 bytes. This would explain why increasing the buffer size did not resolve the problem.

I tested the following delays with the following results.

20ms, consistent errors
30ms, consistent errors
40ms, errors 50% of attempts
50ms, errors 10% of attempts
60ms and over, no errors.

This aligns with the "good practice" of 60ms by the CP2102N datasheet. This also indicates the asynchronous nature of Javascript was likely not allowing any delay between repeated calls of serialWrite().

@jetannenbaum  Since adding more delay did not cause error as well as the change in upload time was barely noticable to a user experience, I went with 100ms to help any other boards that might have a time delay requirement that is slightly more than 60ms. Do you see any problem with a 100ms time delay for the serialWrite() function?

jetannenbaum, 2024-04-12:
Sure, 100ms is fine.  Glad you found a way to resolve the problem. 

Jim

sk8board, 2024-04-14:
Thank you for the teamwork in developing this resolution!

closed 2024-04-14 by sk8board