BIPES/BIPES #253: using components from the pylibsBlobs directory

Romain888, 2025-04-16:
Hello,

I'm interested in the contents of this pylibsBlobs directory.
Unfortunately, after compiling with make offline, I don't see these new elements appear in BIPES.
Is there anything special I need to do?
Sorry for my English.
Thank you.

Romain888, 2025-04-16:
Hello sk8board,

I built my offline version on Linux using the make offline command.
As requested, I tried it on Windows with Cygwin using process #225.
I get the same result.
The pylibsBlobs libraries don't appear in Bipes offline.
I was interested in the PCA9685 library.
Do you have any other ideas?
Thanks for your help.

jetannenbaum, 2025-04-16:
Hi,

   Are you using the offline branch?.  The offline branch has the latest, including the PCA9685 file.

   Jim

jetannenbaum, 2025-04-16:
Conversely, you can use the Pico specific online version: https://bipes.net.br/pico/ui/   By the way the offline branch has only been updated with support for the Pico variants. 

Jim

Romain888, 2025-04-17:
Hello Jim,

Thank you for your answers.
Is there a tutorial that shows me how to transfer a pylibsBlob to an ESP32 or how to "translate" it to update the block_definitions.js, generate_stubs.js, and ESP32.xml files ?
Thanks for your help.

Romain

jetannenbaum, 2025-04-17:
Hello Romain,

   I took a very quick look at the Micropython documentation for the ESP32 (found here: https://docs.micropython.org/en/latest/esp32/quickref.html) since I was concerned about the hardware implementation of the I2C interface.  It seems to have the same implementation as the Pico.  Since that seems to be the case, the only changes needed would be to the ESP32.xml file.  You should be able to copy the applicable sections from the pico.xml file into the esp32.xml file, You will need to update the pin assignments, since the Pico defaults to Pin 0 and 1 for I2C(0) and the ESP32 uses Pins 18 and 19 (again, this is defined in the esp32.xml file).  There is no need to change the index.html, block_definitions.js, gerator_stubs.js or the core.js files, unless you want to add additional peripherals.  Once you update the esp32.xml file, rerun the cygwinOffline.mak file and you should be good to go.  

   Let me know if you need any additional help.  I don't have an ESP32, so I may be limited in answering any issues that come up.

   Jim

Romain888, 2025-04-17:
Jim,

Thank you for your help.
I did the manipulation by taking the PCA9685 part of the Pico to put it in the ESP32
I regenerated the Offline part with cygwin.
While trying to control a servo with the PCA9685, I get the following error:

Traceback (most recent call last):
  File "<stdin>", line 28, in <module>
  File "pca9685.py", line 29, in __init__
  File "pca9685.py", line 38, in reset
  File "pca9685.py", line 32, in _write
OSError: [Errno 19] ENODEV

The address of the PCA9685 hard-coded in the PCA9685.js file was not the right one (I'm on 64 and not 40) 
I do the offline again with cygwin but I get the same thing.
An i2c.scan() tells me that it finds an I2C device on address 64.
Do you have an idea ?

How did you go about transforming the PCA9685.py library ( pylibs ) into a PCA9685.js ( pylibsBlobs ) ?
And how does BIPES know that it has to look for the library in pylibsBlobs and not in pylibs
Thank you and have a good day

Romain

jetannenbaum, 2025-04-18:
Hello,

   40 Hex is 64 Decimal, so that isn't the problem.  Errno 19 ENODEV means that the module couldn't find the I2C device.  Sounds like an issue with the wiring to the pca9685.  The Pico is a 3.3 V device and works correctly.  I think the ESP32 is a 5 V device.  Check that you are connected to the VCC line for the board.  The V+ pin is for 5V for the servos.  The servos draw more current than some micro controllers can supply.  Finally, try the code without any servos connected to the board.  If you are no longer getting errors, then the servos are draining power from the board and you need an external power supply for the V+ pin on the board.  Assuming you need an external 5 V supply, make sure the grounds for the supply and the ESP32 are tied together.

  Converting the pca9685.py library to its pca9685.js equivalent isn't hard.  The file has a var declaration at the top, declaring it to be a Blob, then starting each line with a " and ending the lines with a \n"   BIPES knows to look in the pyLibsBlobs directory because the index_offline.html & index.html file has a pointer to each file that I want to be able to load (look in either html file for pyLibsBlobs and you'll see the script tags that point to the libraries.

Jim

jetannenbaum, 2025-04-18:
Hi Romain,

   You can't load a file via http when you are running offline.  You get a CORs error.  While you can disable the check in your browser, it isn't recommended, due to concerns about scripting attacks.

   I'm familiar with IoTy (Cort has commented here in the past).  Since you are using IoTy and it is working for you, I'm curious how you got it to work without specific blocks to control the PCA9685. 

   Thanks,

    Jim

sk8board, 2025-04-18:
IoTy allows for lower level programming than Bipes.

sk8board, 2025-04-18:
Maybe Romain could share this functional code from IoTy?

jetannenbaum, 2025-04-18:
Since IoTy (as well as Thonny) allows MicroPython code to be entered directly, the code below could be tried to determine what is happening.

MicroPython code to check the board using the Pico (change the I2C pin assignments for the ESP32 board):

from machine import Pin, I2C
from pca9685 import PCA9685

i2c = I2C(0, sda=Pin(0), scl=Pin(1))
print(i2c.scan())

pca9685 = PCA9685(i2c)

pca9685.freq(50)

pca9685.duty(0, 2048)

Jim

Romain888, 2025-04-18:
Hi,

Here are the blocks ( as an attachment) and the Python program ( below ) from IoTy.
The PCA9685 block already exists on IoTy.
This program checks for existing SSIDs, and if SSID xxxx exists, it connects to it; otherwise, it creates an AP.
It then creates a web server that displays a page where I can control a servo via the PCA9685 and the LED on the ESP.
There's an SSD1306 OLED display to display various information.
The web server is limited; it doesn't handle images in HTML pages, nor CSS files, nor Javascript files, so it's a very limited HTML page.
I haven't found a way in IoTy to create my own blocks to make a more "interactive" web server.
That's why I migrated to BIPES, which seems more versatile to me (you can add blocks/functions, and it handles other processors -> ESP boards, Pico boards, Raspberry PI Zero boards).
As for the WebRep manipulation, I'll try it tonight and let you know.
Thank you and have a good day.

Romain

Code Python ->

[image]

import ioty.wifi
import ez_httpd
import machine
import ssd1306
import pca9685
from ioty import pin

ioty_wifi = None
ioty_ap = None
ezhttpd = None
i2c0 = None
ssd1306_i2c = None
i2c1 = None
pca9685_device = None

url = None
query = None
content = None
Type_Wifi = None
ligne_1 = None
ligne_2 = None
ligne_3 = None
Retour_HTTP_Code = None
ligne_4 = None

def Connexion_Wifi():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    ssd1306_i2c.fill(0)
    ssd1306_i2c.text('Config Wifi', 0, 0, 1)
    ssd1306_i2c.text('en cours', 0, 10, 1)
    ssd1306_i2c.show()
    if ioty.wifi.is_present('xxxx'):
        ioty_wifi = ioty.wifi.connect('xxxx', 'zzzz')
    if ioty.wifi.isconnected():
        Type_Wifi = 'Station_Mode'
        ligne_1 = 'SSID :'
        ligne_2 = 'xxxx'
        ligne_3 = 'IP :'
        ligne_4 = ioty.wifi.get_ip()
    else:
        ioty_wifi = ioty.wifi.start_ap('IoTy_Device', '')
        Type_Wifi = 'AP_Mode'
        ligne_1 = 'SSID :'
        ligne_2 = 'IoTy_Device'
        ligne_3 = 'IP :'
        ligne_4 = ioty.wifi.get_ip()
    ssd1306_i2c.fill(0)
    ssd1306_i2c.text((str(ligne_1)), 0, 0, 1)
    ssd1306_i2c.text((str(ligne_2)), 0, 10, 1)
    ssd1306_i2c.text((str(ligne_3)), 0, 20, 1)
    ssd1306_i2c.text((str(ligne_4)), 0, 30, 1)
    ssd1306_i2c.show()

def Initialisation_I2C():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    i2c0 = machine.I2C(0, freq=100000)
    ssd1306_i2c = ssd1306.SSD1306_I2C(128, 64, i2c0, 60)
    ssd1306_i2c.init_display()
    i2c1 = machine.I2C(1, freq=100000)
    pca9685_device = pca9685.PCA9685(i2c1, 64)
    pca9685_device.set_frequency(50)

def fct_Pilote_Servo1_0():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    pca9685_device.servo_deg(0, 0)

def fct_Pilote_Servo1_1():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    pca9685_device.servo_deg(0, 180)

def fct_Pilote_Led1_0():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    pin.digital_write(2, 0)

def fct_Pilote_Led1_1():
    global url, query, content, Type_Wifi, ligne_1, ligne_2, ligne_3, Retour_HTTP_Code, ligne_4
    global ioty_wifi, ioty_ap, ezhttpd, i2c0, ssd1306_i2c, i2c1, pca9685_device
    pin.digital_write(2, 1)

#
# When Started, run the following code
#
Initialisation_I2C()
Connexion_Wifi()
ezhttpd = ez_httpd.HTTPD(address=(ioty.wifi.get_ip()), port=80)
while True:
    if ezhttpd.available():
        url, query, content = ezhttpd.wait_for_connection()
        ezhttpd.send_file('index.html')
        print(url)
        print(query)
        try:
            Retour_HTTP_Code = int(query['fct_Valeur'])
        except:
            pass
        if url == '/fct_Pilote_Servo1/':
            if Retour_HTTP_Code == 0:
                fct_Pilote_Servo1_0()
                ezhttpd.send_bytes()
            elif Retour_HTTP_Code == 1:
                fct_Pilote_Servo1_1()
        elif url == '/fct_Pilote_Led1/':
            if Retour_HTTP_Code == 0:
                fct_Pilote_Led1_0()
            elif Retour_HTTP_Code == 1:
                fct_Pilote_Led1_1()

[image]

jetannenbaum, 2025-04-18:
Hmmm...  It looks like IoTy is using a different library the PCA9685 board.  Note the difference in setting the frequency (freq method vs set_frequency method).  I tried playing with the I2C frequency and that had no effect with my Pico.  I may have to get an ESP32 board to try this, since it should just work with the change you made to the esp32.xml file.

jetannenbaum, 2025-04-18:
Just ordered an ESP32 board from Amazon.  Should be here tomorrow.   I'll let you know what happens after the board arrives and I can test.

Romain888, 2025-04-20:
Hi Jim,

I typed the commands into the WebRepl console and got this:
What's weird is that the line i2c=I2C(......) indicates an error with the i2c variable.
I tried replacing "i2c" with "toto" and it worked.
I continued with the commands and I got the same error message again (Errno 19) with the command line pca9685 = PCA9685(toto).
I'll try switching to I2C 0.
I'll keep you posted.
Have a nice day.

[image]

jetannenbaum, 2025-04-20:
Hi Romain,

I'm starting with a new ESP32/WROOM that I purchased from Amazon.  Had to install the CP210X Universal Windows Driver before I could connect to the ESP32.  Used Thonny to download the latest version of MicroPython (1.25.0).  Type the following in Thonny's code window:
from machine import Pin, I2C

i2c=I2C(1, sda = Pin(26), scl = Pin(25))

print(i2c.scan())

Got back the results of the scan {64, 112]

Closed Thonny and went to BIPES

Replicated the issue.

Went back to Thonny.  Was able to replicate the issue :-)  The SDA and SCL pins are reversed!

Try this:

[image]

Success:

[image]

jetannenbaum, 2025-04-20:
I updated the ESP32.xml file so it properly connects to the board.

Jim

sk8board, 2025-04-20:
Good work Jim. Seemed like a fun problem to support. Sorry I could not help, but I do not have a PCA9685.

jetannenbaum, 2025-04-20:
Thanks.  Not sure I solved it for Romain.  It looks like he was doing all the right things.  Perhaps an older version of MicroPython?  although my seeing the ENODEV when I had the pin definitions reversed seemed like it is the answer.  Guess I'll wait for more feedback before closing the issue.

Romain888, 2025-04-20:
Hi Jim,

I reloaded the new version you made and I'm getting the same error.

Before, I had also reversed I2C 0 (OLED) and 1 (PCA9685), and the same error message.

It's strange because in the PCA9685 block, you set I2C to 0 even though the selected pins correspond to I2C 1 according to the MicroPython table.

[image]

My MicroPython version is 1.23 stable (I used the IoTy programmer).
I'll update to the latest version, 1.25, and retest.
I'll keep you posted.
Have a nice day.

Anyway, thanks for taking the time.

Romain

[image]

[image]

jetannenbaum, 2025-04-20:
Hi Romain,

  My issue was that the wiring to the board was reversed.  While it was correctly configured in BIPES, the wires from the ESP32 to the PCA9685 were not correct.   That was when the error showed up (ENODEV). 

Jim

Romain888, 2025-04-20:
After updating to version 1.25 with Thonny without any error messages (my board is an ESP32 WROOM-32 processor), I get this error message in the Thonny console when I type the command -> from machine import Pin, I2C.
Any ideas?
Thank you very much.

[image]

Romain888, 2025-04-20:
Hourra, it's working now :):)
It was just firmware 1.23 that was causing the problem.
With 1.25, no more problems; I can now control my servos connected to the PCA9685.
Thank you very much for your help and time spent on this.

Now, I need to create a BIPES block for an asynchronous web server (to create an HTML interface with CSS and JavaScript files and thus communicate with the ESP32 from a web browser).
Any ideas?

Have a nice day.

Romain

sk8board, 2025-04-20:
[Here](https://randomnerdtutorials.com/raspberry-pi-pico-w-asynchronous-web-server-micropython/) is a rough starting place.

Looks like you will need make blocks for these modules.

[image]

sk8board, 2025-04-20:
In my opinion,  if you are making a one-off custom program, then IoTy is less work. If you want to contribute to BIPES by adding these modules, then it makes sense to use BIPES.

jetannenbaum, 2025-04-20:
Glad it finally worked.  Not sure what the issue was/is with version 1.23 was, but I'm not going back to that version to further debug :-)

If you look at the online version, in the Pico W microcontroller, under network, you can download a sample of a webserver using BIPES.  It will not give you exactly what you are asking for, but it might get you close enough.

 Jim

sk8board, 2025-04-20:
Thanks Jim. Does BIPES have blocks for WiFi credentials and wlan setup?

[image]

jetannenbaum, 2025-04-20:
Click on Network and Internet.  The blocks are there, since they are common

Romain888, 2025-04-21:
Hi,

I'd like to check if my home's SSID is present when I run a Wi-Fi scan.
If the ESP doesn't find it, it creates an AP; otherwise, it connects to the home's SSID.
There's a block in the IoTy that does this (Wi-Fi SSID "xxx" is present), but not in BIPES.
With BIPES, I managed to output the different SSIDs found during a Wi-Fi scan, but I can't convert the list to text in order to perform a text search for "Maison_AC2200".
I get the following message.
Do you have any ideas?
Thanks and have a nice day.

True
(b'Maison_AC2200', b'b\xa6\xe6}\xcbM', 6, -43, 3, False)
(b'', b'\\\xa6\xe6}\xcbM', 6, -46, 3, False)
(b'Maison_AC2200', b'b\xa6\xe6}\xcbC', 6, -76, 3, False)
(b'', b'\\\xa6\xe6}\xcb\x16', 6, -76, 3, False)
(b'', b'\\\xa6\xe6}\xcbC', 6, -78, 3, False)
(b'Maison_AC2200', b'\xec\x08k\xb8\xd7\xe0', 6, -80, 4, False)
(b'Devolo-CPL', b'\xb8\xbe\xf4\x89\xadW', 1, -92, 3, False)
**Traceback (most recent call last):
  File "<stdin>", line 45, in <module>
TypeError: join expects a list of str/bytes objects consistent with self object**
>>> 

[image]

sk8board, 2025-04-21:
This appears to be a problem of type conversion. You start with a list and want to convert to another type. Why not start with the data type that you want to use or utilize the type you have rather than convert the data type?

jetannenbaum, 2025-04-21:
We are way off topic with this question.  I'll answer the question, but I'm closing the discussion.  Please open a new discussion if you have further questions.

Here are the blocks that should work for you:

[image]

Jim

closed 2025-04-21 by jetannenbaum

sk8board, 2025-04-21:
> This appears to be a problem of type conversion. Why not start with the data type that you want to use or utilize the type you have rather than convert the data type?

Here is an example. I hope this helps with understanding how to determine data type and how to use the data type.

```py
SSID_found = False

data = (b'Maison_AC2200',b'b\xa6\xe6}\xcbM',6,-43,3,False)

print(data)
print(type(data))
print(data[0])
print(type(data[0]))

for i in range(len(data)):
    if data[i] == b'Maison_AC2200':
        SSID_found = True
        
if SSID_found == True:
    print('Target SSID is present')
else:
    print('Target SSID is not available')
```

Result:

(b'Maison_AC2200', b'b\xa6\xe6}\xcbM', 6, -43, 3, False)
<class 'tuple'>
b'Maison_AC2200'
<class 'bytes'>
Target SSID is present

Romain888, 2025-04-21:
Hi,

Okay, Jim's BEEPS blocks are working.
Indeed, this is off-topic from the original topic.
Sorry for the drift.
We can close the thread.
Thanks again to both of you for your help.
Have a nice day.

Romain.